dorssel/usbipd-win #609: 无法映射安卓9008端口

yuxuxin233, 2023-05-27:
[image]
[image]
[image]
[image]

dorssel, 2023-05-27:
Like the warning says: you have an incompatible USB filter driver installed. Either remove it (it is part of https://github.com/daynix/UsbDk), or else first issue `usbipd bind --busid 2-2 --force`.

Duplicate of #264

yuxuxin233, 2023-05-27:
> 

[image]
谢谢

closed 2023-05-28 by dorssel

TaylorTWBrown, 2023-05-30:
Can you provide any context to this issue? I seem to have the same error. Recently upgraded to 3.0.0

yuxuxin233, 2023-05-30:
> 你能提供这个问题的任何背景吗？我似乎有同样的错误。最近升级到 3.0.0

抱歉。我问题复现。没有找到有效方法