dorssel/usbipd-win #610: auto-attach given device instead of busid

terlar, 2023-05-28:
Do you think it would be possible to use device instead of busid when using the auto-attach. I've run into busid changing/flip-flopping. My dist doesn't work with `usbipd wsl attach`, so I have instead implemented an auto-attach via systemd from within my dist using the auto-attach script published with `usbipd-win`.

I had a quick look, but seems most of the operations is bound to `busid` and seems to be no obvious way to use device instead.

dorssel, 2023-05-28:
Correct. Everything is determined by busid (on the Windows side). Even when using `--hardware-id`; that simply translates immediately to the (current) busid.

There are numerous reasons for this (search through other issues). busid is the *only* thing stable on Windows.

Are you saying your busid is not stable ("flip-flopping") on Windows? For the same device plugged into the same physical port? That would be an issue...

terlar, 2023-05-29:
Yes, I experienced this for the same device plugged into the same physical port, I think it was related to me using the computer and running out of battery and then it got `3-1` instead of `4-1`. However now it is back to `4-1` which has been the consistent busid. I'm not sure why it got `3-1` for a while, I was quite surprised. Perhaps it was just a blip, I will change back to `4-1` and pay attention to see if I run into this issue again.

dorssel, 2023-05-29:
Ah. Probably it is the difference between the hub acting as USB2 or USB3. Most act as "two hubs", where the bus number (the first of the busid) is different for all USB2 devices compared to all USB3 devices. So, physically it *is* the same port, but logically it isn't. It makes sense that on low power, the hub refuses to use USB3 (which consumes more power).

Anyways, you can `bind` both, and have simultaneous `auto-attach` scripts running for both busids. Then it will just pick whichever it is.