dorssel/usbipd-win #618: WSL2 + Ubuntu 20.04 + Serial COM PL2303 not associate on  /dev/ttyUSB

pablitonadal, 2023-06-02:
hi guys,

My configuration is WSL2 + Ubuntu 20.04

I followed this steps on WIKI WSL support, I tried with "WSL setup" and "wsl-convenience-commands" guide but in both case the result is same. I don't see serial port in /dev/* .

[image]

[image]

[image]

[image]

[image]

[image]

**how can I solve this problem?**

Thanks.

dorssel, 2023-06-02:
Duplicate, with answer here: https://github.com/dorssel/usbipd-win/issues/426#issuecomment-1203257603.

dorssel, 2023-06-02:
You need to enable the right driver. See: https://www.kernelconfig.io/config_usb_serial_pl2303

pablitonadal, 2023-06-02:
Thank you very much @dorssel

I followed step by wiki  (https://github.com/dorssel/usbipd-win/wiki/WSL-support) guide and in particular the part "Building your own USB/IP enabled WSL 2 kernel"
When i arrived in step "sudo make menuconfig" I follewed your link instruction that you shared and i had select "USB Prolific 2303 Single Port Serial Driver"

[image]

When I fineshed guide steps I tried share serial com and ubuntu 20.04 it mapped on ttyUSB0

[image]

Has worked for me

thanks.

closed 2023-06-02 by pablitonadal