dorssel/usbipd-win #629: Incorrect configuration path in wiki (WSL support)

o-alquimista, 2023-06-09:
From https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel:

> These are the necessary additional features in menuconfig.
> Device Drivers -> USB Support
> Device Drivers -> USB Support -> USB announce new devices
> Device Drivers -> USB Support -> USB Modem (CDC ACM) support
> Device Drivers -> USB Support -> USB/IP
> Device Drivers -> USB Support -> USB/IP -> VHCI HCD
> Device Drivers -> USB Support -> USB/IP -> Debug messages for USB/IP
> Device Drivers -> USB Serial Converter Support
> Device Drivers -> USB Serial Converter Support -> USB FTDI Single port Serial Driver

The path to `USB Serial Converter Support` has changed. It's under `USB Support` now.

```
Device Drivers -> USB Support -> USB Serial Converter Support
Device Drivers -> USB Support -> USB Serial Converter Support -> USB FTDI Single port Serial Driver
```

dorssel, 2023-06-12:
Thanks for the report!

I will leave this issue open for others to find. It may take a bit before I finally get around to updating the docs (I need to overhaul the entre thing, there are more issues...)

ejhuff, 2023-12-18:
For custom kernel building, the make install is unnecessary and confusing. It turns out that wsl never uses the kernels in /boot, while make install puts them there. 

sudo make -j 8 && sudo make modules_install -j 8 && sudo make install -j 8

should omit the make install and use nproc:

np=$(nproc) #counts which processors are allocated to this process
sudo make -j $np && sudo make modules_install -j $np && sudo make install -j $np

Official custom kernel for wsl instructions here: 
https://learn.microsoft.com/en-us/community/content/wsl-user-msft-kernel-v6

dorssel, 2023-12-18:
@ejhuff 
Thanks for the improvements. The WSL kernel docs in wiki are rather old; they actually pre-date the USBIP-capable kernels from years ago. At some point, I will replace the wiki with a reference to the Microsoft docs (now that they exist).

About the modules: the WSL kernel is monolithic; there are no modules to find.

timsu92, 2024-02-17:
> The path to USB Serial Converter Support has changed. It's under USB Support now.

This is correct. However, at the time I'm building this right now, the options mentioned above are already turned on in WSL2.