dorssel/usbipd-win #630: Missing kernel build dependency: bc

o-alquimista, 2023-06-09:
In addition to the packages `build-essential flex bison libssl-dev libelf-dev libncurses-dev autoconf libudev-dev libtool`, I also had to install the package `bc`, otherwise the build would fail very early with `bc: not found`.

I'm not sure if this issue belongs here or at https://github.com/microsoft/WSL2-Linux-Kernel

dorssel, 2023-06-10:
Did you enable any special drivers? That may be the cause, and would be covered by https://github.com/dorssel/usbipd-win/issues/429 (although not exactly the same, but close enough). Different drivers depend on different dependencies. The packages in the wiki should be sufficient to compile the default WSL kernel.

o-alquimista, 2023-06-10:
Yes, I needed to enable USB storage (for USB sticks). That explains it. Thanks.

closed 2023-06-10 by o-alquimista