dorssel/usbipd-win #635: usbipd wsl attach not working anymore

ThierryCam, 2023-06-14:
Not sure what break my setup, maybe a windows update or WSL update, but it use to work with same setup (Windows 10 PC and USB device) and now I cannot attach anymore.

Running with upbipd server, I get following log (note that I don't have a D:\ disk mounted on my computer) : 

```
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 308
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 167
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 270
         at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
```
Any idea what could be wrong ?

ThierryCam, 2023-06-14:
Power cycle the PC (not just a restart) seems to have fixed the issue. Really strange.

ThierryCam, 2023-06-19:
Still got same issue again, an power cycle doesn't solve this time. 

Note that on the "Device Manager", I have the "USBIP Shared Device" reporting an error : 
```
Windows cannot load the device driver for this hardware. The driver may be corrupted or missing. (Code 39)
The specified page protection was not valid.

```
Look like an issue with VBoxUSB. But no more indication of what could be wrong.

dorssel, 2023-06-19:
What version of `usbipd-win` are you using? Sounds like the driver wasn't installed quite correctly. Maybe try a repair install or re-install. Do you run any special anti-virus? Have you got VirtualBox itself installed (`usbipd-win` works better without it...)?

ThierryCam, 2023-06-20:
3.0.0+361.Branch.master.Sha.89d4b327c99a4eecc9b583e1843a428aae3c9270

I don't have virtual box installed.

One hint: it seems that it always works correctly if I run the command as an administrator. But randomly if not administrator.

ThierryCam, 2023-06-23:
Thanks. The "run as administrator" seems to work reliably.

closed 2023-06-23 by ThierryCam