dorssel/usbipd-win #637: Not able to attach devices using 'attach'

Josh23Og, 2023-06-20:
I recently installed usbipd and followed the instructions but when I try to attach one of the devices using 
`usbipd wsl attach --busid=2-1`
I get the error 
`usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions.`
Most of the commands using wsl and usbipd work, it only throws up the error when I try to attach a device

dorssel, 2023-06-21:
You need to set up the *client* (WSL Linux) side `usbip` tooling properly.

base1129, 2023-06-21:
I'm encountering the same issue. How do you properly set up the tooling?

dorssel, 2023-06-21:
What is the error you get when you execute the instructions in the wiki?

RBE-Motion, 2023-06-23:
I'm having the same exact issue today, trying to attach my USB device

dorssel, 2023-09-03:
Stale

closed 2023-09-03 by dorssel