dorssel/usbipd-win #643: Reinstall of usbipd-win Failing with Error Status 1603

mmclemens, 2023-06-23:
Previously was having trouble with attach/detach of a USB device causing blue screen crashes, possibly due to having done a forced bind where it looked to have been in a shared state. Attempted to normalize system by uninstalling USBIP Shared Device and usbipd-win software, rebooted, and the tried to reinstall usbipd-win software without success. MsiInstaller Event Viewer shows error Status 1603. Any ideas for a work around to get usbipd-win software reinstalled?

dorssel, 2023-06-23:
Try another reboot before installing. And get a log of the installation process by running:
```
msiexec /package usbipd-win_3.0.0.msi /log install-log.txt
```

mmclemens, 2023-06-23:
Thank you for your quick reply. Rebooted machine and executed "msiexec /package usbipd-win_3.0.0.msi /log install-log.txt" within Administrator Command Prompt, still without success and have install-log.txt attached.
[install-log.txt](https://github.com/dorssel/usbipd-win/files/11849445/install-log.txt)

dorssel, 2023-06-23:
Same problem as here https://github.com/dorssel/usbipd-win/issues/313#issuecomment-1072899440

Something goes wrong when installing the driver. See suggestions in the mentioned comment. Besides driver database corruption, it could also be company policies or third party endpoint protection software preventing driver installation.

mmclemens, 2023-06-23:
Ah, thanks, I've been working through some of your suggestions. I downloaded a new usbipd-win, still had same rollback behavior. No "vbox" string in "pnputil /enum-drivers". Anything else besides running "chkdsk", maybe something in regedit?
[enum-drivers.txt](https://github.com/dorssel/usbipd-win/files/11849678/enum-drivers.txt)

dorssel, 2023-06-23:
Since Windows PnP "believes" that no VBoxUSB driver is currently installed, maybe you need to cleanup some left-over files in C:\Windows\System32\drivers. You can safely remove VBoxUSB.sys (and possibly VBoxUSBMon.sys). Those shouldn't exist, so if they do: remove them.

mmclemens, 2023-06-23:
Thanks for your reply. I have deleted the VBox files from C:\Windows\System32\drivers, rebooted, and reran usbipd-win_3.0.0.msi, the captured log looks to be similar to the past logs.

```
DIFXAPP: INFO:   Could not open file C:\Windows\System32\DriverStore\FileRepository\vboxusb.inf_amd64_e2b97b308ad75564\vboxusb.inf. (Error code 0x3: The system cannot find the path specified.)
DIFXAPP: ERROR:  PnP Install failed. (Error code 0x3EE: The volume for a file has been externally altered so that the opened file is no longer valid.)
```

I had also ran chdsk, but it seemed to run relatively quick. Any other ideas?

dorssel, 2023-06-24:
Remove all directories `C:\Windows\System32\DriverStore\FileRepository\vboxusb.inf_amd64_*\`

mmclemens, 2023-06-24:
Still no luck. No sign of any C:\Windows\System32\DriverStore\FileRepository\vboxusb directory. Rebooted, and retired but installation still unsuccessful.

[image]

[install-log04.txt](https://github.com/dorssel/usbipd-win/files/11854035/install-log04.txt)

mmclemens, 2023-06-24:
Thanks for your reply, and do apologize for having caused this condition where a reinstallation is failing. The usbipd-win had been working very well the past few months, but recently I had put the USB device in a state where the attach was causing a blue screen crash. So, I had uninstalled the USB device driver and usbipd-win to start from scratch but having no luck with the usbipd-win reinstallation.

I ran sfc and cleanmgr utilities, rebooted, and attempted another reinstall without success. A diff with the prior logs looks to be the same fault.

[install-log05.txt](https://github.com/dorssel/usbipd-win/files/11857238/install-log05.txt)

dorssel, 2023-06-24:
I found this: worth a try...

https://forums.virtualbox.org/viewtopic.php?f=2&t=102401

Download all files from this directory (or use 7zip to unpack the MSI): https://github.com/dorssel/usbipd-win/tree/v3.0.0/Drivers/VBoxUSB

Right-click on VBoxUSB.inf and select Install. From the discussion above it seems to help to do a one-time manual driver installation, after which the installer *should* work again.

mmclemens, 2023-06-24:
Hello Frans, thanks so much for continuing to support me with the usbipd-win reinstallation effort. I think we are on the same track. I had found an article earlier today https://www.exefiles.com/en/inf/vboxusb-inf/. I attempted to recreate the associated vboxusb.inf directory and had run into a permissions issue. I went to the FileRepository directory, selected the Security tab, and the "Everyone" Group to allow full control and modify, It had a dialogue popup "Error Applying Security" for the containing files: "Failed to enumerate objects in the container. Access id denied." I cancelled the operation FileRepository permissions were modified enough to allow making the vboxusb.inf directory, I then copied the downloaded VBoxUSB.inf (VirtualBox 6.0.14) into the vboxusb.inf directory.

```
C:>cd C:\Windows\System32\DriverStore\FileRepository
C:>mkdir vboxusb.inf_amd64_e2b97b308ad75564
Access is denied.
C:>mkdir vboxusb.inf_amd64_e2b97b308ad75564
C:>
```
I was able to run a successful usbipd-win reinstallation and noted it looked to be unhappy with the state of the VBoxUSB.inf (VirtualBox 6.0.14) file: is this maybe a concern?
```
DIFXAPP: INFO:   Looking for Model Section ...
DIFXAPP: INFO:   No matching devices found in INF "C:\Windows\System32\DriverStore\FileRepository\vboxusb.inf_amd64_e2b97b308ad75564\vboxusb.inf" on the Machine.
DIFXAPP: INFO:   No drivers installed. No devices found that match driver(s) contained in 'C:\Windows\System32\DriverStore\FileRepository\vboxusb.inf_amd64_e2b97b308ad75564\vboxusb.inf'.
DIFXAPP: INFO:   RETURN: DriverPackageInstallW  (0xE000020B)
```
[install-log06.txt](https://github.com/dorssel/usbipd-win/files/11857816/install-log06.txt)
I was able to launch an usbipd command:
```
PS C:\Windows\system32> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
2-3    0c45:6720  Integrated Webcam                                             Not attached
2-5    0a5c:5842  Dell ControlVault w/o Fingerprint Sensor, Microsoft Usbcc...  Not attached
2-9    046d:081b  USB Video Device, USB Audio Device                            Not attached
2-11   056e:00f9  USB Input Device                                              Not attached
2-14   8087:0032  Intel(R) Wireless Bluetooth(R)                                Not attached
3-4    0bda:8153  Realtek USB GbE Family Controller                             Not attached
7-5    413c:b06e  USB Input Device                                              Not attached
8-2    1366:0101  J-Link driver                                                 Not attached
8-3    04d8:ffee  USB Serial Device (COM2)                                      Not attached
9-5    413c:b06f  USB Input Device                                              Not attached
10-1   0483:3748  STM32 STLink                                                  Not attached
10-2   0403:6010  USB Serial Converter A, USB Serial Converter B                Not attached
10-3   0403:6015  USB Serial Converter                                          Not attached
10-4   0403:6001  USB Serial Converter                                          Not attached
```
I wonder if it may have just been a permissions issue causing the reinstallation to fail? So, my next step is to re-enable the USB device to see how it fares. I will keep you posted.

mmclemens, 2023-06-24:
Just copied your VBoxUSB.inf file from https://github.com/dorssel/usbipd-win/tree/v3.0.0/Drivers/VBoxUSB and did a reinstallation usbipd-win (repair) and the log looks to be happier:
```
DIFXAPP: INFO:   Attempting to repair driver with INF 'C:\Windows\System32\DriverStore\FileRepository\vboxusb.inf_amd64_e2b97b308ad75564\VBoxUSB.inf'.
DIFXAPP: INFO:   'C:\Windows\System32\DriverStore\FileRepository\vboxusb.inf_amd64_e2b97b308ad75564\VBoxUSB.inf' matches OEM Inf 'oem4.inf'.
DIFXAPP: SUCCESS:Install completed
DIFXAPP: INFO:   RETURN: DriverPackageInstallW  (0x0)
```
[install-log07.txt](https://github.com/dorssel/usbipd-win/files/11857865/install-log07.txt)
Still verifying USB device operation...

mmclemens, 2023-06-24:
Seeing "USBIP Shared Device" in "Device Manager's "Universal Serial Bus controllers" after the "usbipd wsl attach" command! With the VBoxUSB.inf (VirtualBox 6.0.14) file, Device Manager categorized it under "Virtual Box" and had reported: "usbip: error: Attach Request for 8-1 failed - Device in error state".

Using the your proper VBoxUSB.inf file, I am observing: "8-1    20a0:4230  Microsoft Usbccid Smartcard Reader (WUDF) Attached - Ubuntu". and WSL lsusb shows: "Bus 001 Device 002: ID 20a0:4230 Clay Logic Nitrokey HSM".

Thank you so much for your help and patience in getting me back to a good state. I think this issue can be closed.

dorssel, 2023-06-24:
> I wonder if it may have just been a permissions issue causing the reinstallation to fail?

Yes, that sounds plausible. Anyways. Good you're back in business. I'll close this one.

closed 2023-06-24 by dorssel

kodonnell, 2024-04-02:
FWIW I spent an hour on this for 4.0.0 -> 4.1.0. For me, it appeared to hang while validating, so I killed it - after that I couldn't uninstall 4.0.0 with the 1603 issue. What eventually worked was reinstalling 4.0.0 and repairing it.