dorssel/usbipd-win #646: linux-tools-5.15.90.1-microsoft-standard-WSL2 demanded, but nonexistent

ACleverDisguise, 2023-06-29:
This is a WSL2 spin-up of a Ubuntu 22.04 instance:

```
username@DESKTOP-HJV57MC:~$ uname -a
Linux DESKTOP-HJV57MC 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
username@DESKTOP-HJV57MC:~$
```

Launched from Windows 10:

```
Microsoft Windows [Version 10.0.19045.3086]
(c) Microsoft Corporation. All rights reserved.

C:\Users\UserName>
```

After following the instructions at the top of https://github.com/dorssel/usbipd-win/wiki/WSL-support I have a usbip command, but it does this:

```
username@DESKTOP-HJV57MC:~$ usbip
WARNING: usbip not found for kernel 5.15.90.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.15.90.1-microsoft-standard-WSL2
    linux-cloud-tools-5.15.90.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2
username@DESKTOP-HJV57MC:~$
```

Trying to install any of the named packages results in outcomes like this:

```
username@DESKTOP-HJV57MC:~$ sudo apt install linux-tools-5.15.90.1-microsoft-standard-WSL2 linux-tools-standard-WSL2
[sudo] password for username:
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-tools-5.15.90.1-microsoft-standard-WSL2
E: Couldn't find any package by glob 'linux-tools-5.15.90.1-microsoft-standard-WSL2'
E: Unable to locate package linux-tools-standard-WSL2
username@DESKTOP-HJV57MC:~$
```

Sure enough, searching for anything with `WSL` in the package name results in:

```
username@DESKTOP-HJV57MC:~$ apt list *WSL*
Listing... Done
fusiondirectory-plugin-newsletter-schema/jammy 1.3-4build1 all
fusiondirectory-plugin-newsletter/jammy 1.3-4build1 all
libwslay-dev/jammy 1.1.1-3 amd64
libwslay-doc/jammy 1.1.1-3 all
libwslay1/jammy 1.1.1-3 amd64
python3-owslib/jammy 0.25.0-1 all
ubuntu-wsl/jammy,now 1.481 amd64 [installed]
wsl-setup/jammy,now 0.2 amd64 [installed]
wsl/jammy 0.2.1-3 all
wslu/jammy,now 3.2.3-0ubuntu3 amd64 [installed]
username@DESKTOP-HJV57MC:~$
```

Googling for those package names results in nothing helpful for finding and installing them.

I'm at the end of what I can figure out from here.  Any suggestions for finding the mysterious missing packages?

dorssel, 2023-06-29:
What is the output of the instructions in https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools?

RedTail72, 2023-07-16:
@ACleverDisguise Did you run the ```sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20``` command from the Wiki?  What is the output of `which usbip`?  If it is pointing to `/usr/bin/usbip`, that is your problem.  Running the update-alternatives command should fix that.

jtegtmeier, 2023-07-17:
After compiling and using a custom kernel, I was able to get past this.
https://github.com/dorssel/usbipd-win/discussions/127

umie0128, 2023-07-20:
@ACleverDisguise  
Have you solved it yet？

umie0128, 2023-07-20:
@dorssel @RedTail72 
[image]

dorssel, 2023-07-20:
You're using a distro with kernel 4.15 as default. A really old Ubuntu 18 maybe?. Try installing the hwe toolset for that distro. Search through comments on other issues. It has been solved before.

RedTail72, 2023-07-20:
@umie0128 I just tried this from a fresh Ubuntu 18.04 install for WSL and I was able to run this.  You may need to run `sudo apt upgrade` and `sudo apt update` to get the latest versions of the tools.

[image]

The complete list of commands I ran from my fresh installed:
```
sudo su -
apt upgrade
apt update
apt install linux-tools-virtual hwdata
update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20
lsusb
```
** NOTE: I ran `lsusb` after attaching the device to this distro from the admin powershell window.

shrutichakraborty, 2023-11-22:
> @umie0128 I just tried this from a fresh Ubuntu 18.04 install for WSL and I was able to run this. You may need to run `sudo apt upgrade` and `sudo apt update` to get the latest versions of the [...]

Hi all! I am trying to access my laptop's inbuilt camera from WSL ubuntu. I have followed all the steps above and in the end I have : 

[image]

[image]

So it seems that I can see my camera on executing` lsusb ` from within ubuntu. I then tried a basic opnCV videoCapure code to test the camera

```
import cv2
cap = cv2.VideoCapture(0)
while True: 
    ret, frame = cap.read()
    if not ret:
        continue
    cv2.imshow('usb cam test', frame)
    if cv2.waitKey(1) & 0xFF == ord('q'):
        break

cap.release()
cv2.destroyAllWindows()
```
However, I get the following error/warning messages : 

[image]

[image]

Any help will be highly appreciated!!

dorssel, 2023-11-22:
@shrutichakraborty 
You're hijacking this issue. Yours is not related at all. Please file a new one instead. (Hint: look through older issues about video devices. You're missing driver in the stock kernel; you need to build your own...)

dorssel, 2023-12-06:
4.0.0 has been released and installing client-side tooling is no longer necessary.