dorssel/usbipd-win #647: Unable to attach Sierra Wireless modem to WSL2 (can't set config #1, error -110)

mfe-dev, 2023-06-29:
Hi,

Thanks for your work on this !

I'm facing an issue when trying to attach my WP7607 devkit (from Sierra Wireless) to WSL2. From dmesg, it seems that there was an issue with interface number ... 
```
[103150.752898] vhci_hcd vhci_hcd.0: pdev(0) rhport(1) sockfd(3)
[103150.752902] vhci_hcd vhci_hcd.0: devid(65540) speed(3) speed_str(high-speed)
[103150.752926] vhci_hcd vhci_hcd.0: Device attached
[103151.096850] usb 1-2: new high-speed USB device number 16 using vhci_hcd
[103151.246712] usb 1-2: SetAddress Request (16) to port 1
[103151.293817] usb 1-2: config 1 has an invalid interface number: 8 but max is 4
[103151.293820] usb 1-2: config 1 has an invalid interface number: 19 but max is 4
[103151.293821] usb 1-2: config 1 has an invalid interface number: 20 but max is 4
[103151.293821] usb 1-2: config 1 has an invalid interface number: 20 but max is 4
[103151.293822] usb 1-2: config 1 has no interface number 1
[103151.293822] usb 1-2: config 1 has no interface number 2
[103151.293822] usb 1-2: config 1 has no interface number 4
[103151.303594] usb 1-2: New USB device found, idVendor=1199, idProduct=68c0, bcdDevice= 4.14
[103151.303598] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[103151.303599] usb 1-2: Product: Sierra Wireless WP7607
[103151.303599] usb 1-2: Manufacturer: Sierra Wireless, Incorporated
[103151.303600] usb 1-2: SerialNumber: 29d8277a
[103151.318458] vhci_hcd: connection closed
[103151.318564] vhci_hcd: stop threads
[103151.318566] vhci_hcd: release socket
[103151.318569] vhci_hcd: disconnect device
[103156.376736] usb 1-2: can't set config #1, error -110
[103156.377304] usb 1-2: USB disconnect, device number 16
```
I attached the usbview from Windows

Also, I've found [that](https://forum.sierrawireless.com/t/problem-with-the-linux-driver-of-mc7355/8178) on Sierra Wireless forum. It's related to another device, but it seems to be the same issue on Linux side. I tried to add this udev rules (after changing the pid) but it didn't fix anything

Let me know if there is a solution to access this device from WSL2

BR,
Mathieu F.

[usbview.txt](https://github.com/dorssel/usbipd-win/files/11906060/usbview.txt)

dorssel, 2023-06-29:
Please follow https://github.com/dorssel/usbipd-win/wiki/Troubleshooting to get full logging on the `usbipd-win` service and include a USB capture as well.

dorssel, 2023-06-29:
The device resets after `SET_CONFIGURATION: 1`, since it immediately reports:

> Unbind or unplug while attached

That's not what it is supposed to do. I'm afraid the device is incompatible with the USB spec; `usbipd-win` really requires to pass SET_CONFIGURATION, as it also is used by the VirtualBox driver to set up all the interfaces and pipes.

mfe-dev, 2023-06-30:
That's what I was afraid of ... Thanks for your analysis anyway 🙏🏻