dorssel/usbipd-win #656: Unknown USB Device (Set Address Failed) occurs when using USB3.0 Device

jongchoon, 2023-07-20:
Hello.

The device used is a USB3.0 Basler Camera. It works normally only when connected to USB3.0.
An error occurs in the USB3.0 connection state.
I accidentally connected it with USB 2.0, and then I confirmed that it was "attached" normally.

The version I am using is as follows.
```
Windows 10 22H2 (OS build 19045.3208)
WSL 5.15.90.1-microsoft-standard-WSL2
USBIPD 3.0.0+361.Branch.master.Sha.89d4b327c99a4eecc9b583e1843a428aae3c9270
```

The test results for Case 1 &2 are as follows. 

Case1. USB2.0 connection

```
PS C:\Users\jochkim> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-5    046d:c52b  Logitech USB Input Device, USB Device                      	Not attached
2-1    0403:6015  USB Serial Converter                                          Not attached
2-2    0403:6015  USB Serial Converter                                          Not attached
2-4    2676:ba02  Basler ace USB3 Vision Camera                                 Not attached

PS C:\Users\jochkim> usbipd wsl attach -b 2-1
PS C:\Users\jochkim> usbipd wsl attach -b 2-2
PS C:\Users\jochkim> usbipd wsl attach -b 2-4
PS C:\Users\jochkim> usbipd wsl list
BUSID  VID:PID    DEVICE                                                        STATE
1-5    046d:c52b  Logitech USB Input Device, USB Device                      	Not attached
2-1    0403:6015  USB Serial Converter                                          Attached - Ubuntu-20.04
2-2    0403:6015  USB Serial Converter                                          Attached - Ubuntu-20.04
2-4    2676:ba02  Basler ace USB3 Vision Camera                                 Attached - Ubuntu-20.04
```

```
jochkim@DESKTOP-HGQRKCT:~/Downloads$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 2676:ba02 Basler AG ace
Bus 001 Device 003: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 002: ID 0403:6015 Future Technology Devices International, Ltd Bridge(I2C/SPI/UART/FIFO)
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
PS C:\Users\jochkim> usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      3.0.0+361.Branch.master.Sha.89d4b327c99a4eecc9b583e1843a428aae3c9270
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveInterval = 500 ms
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveTime = 10000 ms
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Users\jochkim
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 192.168.248.104
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
[... 189 lines cut ...]
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 270
         at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
dbug: Usbipd.Server[1000]
      connection close: Unable to find the specified file.
dbug: Usbipd.Server[1000]
      connection closed
info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
dbug: Microsoft.Extensions.Hosting.Internal.Host[3]
      Hosting stopping
info: Usbipd.ConnectedClient[2]
      Client 192.168.248.104 released device at 3-2 (USB\VID_0403&PID_6015\DK0CFRQK).
dbug: Microsoft.Extensions.Hosting.Internal.Host[4]
      Hosting stopped
info: Usbipd.ConnectedClient[2]
      Client 192.168.248.104 released device at 3-1 (USB\VID_0403&PID_6015\DK0CGG3K).
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
dbug: Usbipd.Server[1000]
      connection close: The operation was canceled.
dbug: Usbipd.Server[1000]
      connection close: The operation was canceled.
```

dorssel, 2023-09-03:
This may have been fixed with #677. Could you try the installer from the latest master build (not released yet)? It is at https://github.com/dorssel/usbipd-win/actions/runs/6017037658.

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel