dorssel/usbipd-win #657: Easy-fix to usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.

MocaCDev, 2023-07-20:
Open _Control Panel_ on Windows, go to _System and Security_ then click on _Windows Defender Firewall_.

Click on _Advanced settings_ on the left panel, and this will open up a window:
[image]

Once this opens, go to _Inbound Rules_ and find _usbipd_, which will be displayed in the middle panel of the window:
[image]

Click on it, then click on _Properties_:
[image]

Go to the _Protocols and Ports_ tab, and change _Local port_ from _Specific Ports_ to _All Ports_.
Try using the tool now, it should work just fine!

closed 2023-07-20 by MocaCDev

zzzh, 2023-08-20:
Tried, can't find usbipd in the list

ZC-eto, 2025-01-07:
It still doesn’t work. Even after modifying the rule, it still doesn’t work. I turned off the Windows Firewall, turned off the security software, and the VPN still had the same error. This problem will occur as long as WSL uses virtioproxy network mode.

dhruvil1607, 2026-04-01:
I solved by shifting WSL Ubuntu 22.04 to `mirrored` mode.
Steps to change network mode of WSL.

1. Open the .wslconfig file:
Open PowerShell or Command Prompt and run the following command to open the configuration file in Notepad. This file is located in your Windows user profile folder (%UserProfile%):
```
notepad $env:USERPROFILE\.wslconfig
```
If the file does not exist, it will be created when you save it.

2. Add the configuration setting:
Add the following lines to the .wslconfig file. If a [wsl2] section already exists, just add the networkingMode line under it:
```
[wsl2]
networkingMode=mirrored
```
Make sure there are no spaces around the equals sign (=) for correct parsing. You can also enable related features like dnsTunneling and autoProxy for better compatibility, but they are often on by default with recent WSL versions:

```
[wsl2]
networkingMode=mirrored
dnsTunneling=true
firewall=true
autoProxy=true
```

3. Save and close the file.

4. Restart WSL:
Shut down all running WSL instances for the changes to take effect by running this command in PowerShell or Command Prompt: In powershell
```
wsl --shutdown
```
Wait a few seconds, then launch a WSL distribution again (e.g., by opening a Ubuntu terminal).

5. Verify the network model
```
wslinfo --networking-mode
```