dorssel/usbipd-win #659: Timing Issue When Attaching Certain Devices

RedTail72, 2023-07-20:
When testing with 2 SanDisk USB storage devices, they fail to attach with error message `usbip: error: Attach Request for 3-13 failed - Device in error state`.

Cloning down the project locally, I was able to run it locally in debug mode.  On my first run, I set a breakpoint in [GetVBoxDevice](https://github.com/dorssel/usbipd-win/blob/0726f7f08e5a542ace4d5addcf3805ccf7e11e50/Usbipd/ConfigurationManager.cs#L292) and found deviceInterfaces was empty (which explained the error). I decided to step farther back and set a breakpoint in [Get_Device_Interface_List](https://github.com/dorssel/usbipd-win/blob/0726f7f08e5a542ace4d5addcf3805ccf7e11e50/Usbipd/ConfigurationManager.cs#L48). When I set it there, I found the list was populated for each call into the method. It was a little slower, so I'm guessing it has something to do with the device driver being replaced.

Opening as a separate issue for easier tracking, based on the conversation from  https://github.com/dorssel/usbipd-win/issues/425#issuecomment-1637197756.

Output from server logs:
```
dbug: Usbipd.Server[1000]
      new connection from 192.168.29.148
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 192.168.29.148
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 308
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 167
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 270
         at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
dbug: Usbipd.Server[1000]
      connection close: Unable to find the specified file.
dbug: Usbipd.Server[1000]
      connection closed
```

dorssel, 2023-08-14:
@RedTail72 
Received the device in good order. Will keep you updated as soon as I have some results. Thanks!

dorssel, 2023-08-16:
@RedTail72 
Could you test your (other) devices with the build artifact from #677 ? The installer is here: https://github.com/dorssel/usbipd-win/actions/runs/5879219313?pr=677. Note that the installer is not signed, but the drivers are.

It fixes the issue (which was very reproduceable) with the SanDisk Ultra you sent me.

dorssel, 2023-08-17:
@thijsputman 
It *could* be, but I don't think so. The SanDisk issue is that the *device* locks up, not the driver. In your YubiKey case it is the driver that fails to load; the device is still responsive as in "it appears in the device list"  (right?). With SanDisk, the entire device disappears until replug.

thijsputman, 2023-08-18:
Correct; the device itself doesn't appear to lock up in my case – I'll keep an eye on this issue nonetheless.

RedTail72, 2023-08-18:
I will try it out this weekend and let you know.

RedTail72, 2023-08-21:
So I can't read as I somehow missed your link to the installer.  I did
switch over to your branch and ran it without breakpoints.  It worked fine,
just like the other devices.  I'll test out the installer tomorrow and let
you know.

RedTail72, 2023-08-27:
> @RedTail72 Could you test your (other) devices with the build artifact from #677 ? The installer is here: https://github.com/dorssel/usbipd-win/actions/runs/5879219313?pr=677. Note that the [...]

@dorssel  I have tested the build artifact you linked to and was able to mount the SanDisk drive without any issues.  I confirmed expected confirmed behavior with another previously validated USB drive as well.

From my perspective, this looks good and resolves this issue.

closed 2023-08-28 by dorssel

dorssel, 2023-08-28:
@RedTail72 
I've merged the change. GitHub auto-closed the issue. You can use the dev version until I make a new release.

Do you have a return address for the SanDisk, or do you want me to keep it for regression testing?

RedTail72, 2023-08-28:
Keep it for regression testing or whatever else you see fit.  Just glad I
could help you out.

PR #677 (2023-08-28): Fix port reset timing
Fixes #659