dorssel/usbipd-win #663: Unable to attach USB devices (usbipd: error: Failed to attach device with busid <busid>)

Animex77, 2023-07-25:
Until today usbipd works well except it shows that usbip in Linux is not installed properly few months ago and I managed to fix it. That problem occurred again today, but after I solved that problem, a new issue happened:
As I use ```attach``` command, terminal shows:
```
PS C:\Users\84766> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-3    0b05:19b6  USB 输入设备                                                  Not shared
2-5    17cc:1830  Komplete Audio 1, Komplete Audio 1 DFU                        Not shared
2-6    046d:c539  USB 输入设备                                                  Not shared
2-10   8087:0033  英特尔(R) 无线 Bluetooth(R)                                   Not shared
2-13   8086:0b5c  Intel(R) RealSense(TM) Depth Camera 455  Depth, Intel(R) ...  Not shared

Persisted:
GUID                                  DEVICE

PS C:\Users\84766> usbipd wsl attach --busid 2-13
usbip: error: Attach Request for 2-13 failed - Device in error state

usbipd: error: Failed to attach device with busid '2-13'.
``` 

and using ```list```  command after I tried ```attach``` , the terminal shows:
``` PS C:\Users\84766> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-3    0b05:19b6  USB 输入设备                                                  Not shared
2-5    17cc:1830  Komplete Audio 1, Komplete Audio 1 DFU                        Not shared
2-6    046d:c539  USB 输入设备                                                  Not shared
2-10   8087:0033  英特尔(R) 无线 Bluetooth(R)                                   Not shared

Persisted:
GUID                                  DEVICE
6c4d31c9-4464-4a86-a8d7-bc2af5470281  Intel(R) RealSense(TM) Depth Camera 455  Depth, Intel(R) ...l``` 
This device was able to attach earlier. Also, I tried to attach other USB devices to WSL using usbipd and usbip, but I still have this error or ```Device busy (exported)```. I hope to get some suggentions on solving this problem, thanks!

SchnTgaiSpock, 2023-07-25:
i have the same issue conecting an android phone, broke on 3.1.0 but fine on 2.4.1

Animex77, 2023-07-26:
I actually use 2.4.1 at first and my issue came out... Then I tried 3.1.0 and 3.0.0. I installed 2.4.1 again and it still don't work so I bind 2-10 force, reboot, and it seems it works again even the busID of the device changed to 2-2. I would test the device that failed to attach in my earlier post later.

dorssel, 2023-08-18:
> Device busy (exported)

This means that Windows (or some software on the host) is using the device. You may have to use `bind --force` and a reboot. That will prevent Windows from using the device at all, after which you should be able to attach from the client.

closed 2023-09-03 by dorssel

lihakuii, 2024-09-13:
When I use the Xbox controller, although I` bind --force`, I still can't connect when I try to `attach` it after restarting the computer, .Even though it shows a successful connection in a timely manner. However, during verification, I still found that the actual connection to WSL was not successful。
[image]
[image]