dorssel/usbipd-win #665: Can't install DYMO printer in wsl

baobabprince, 2023-07-27:
I have dymo 450 printer that connected to my computer, and I try to use it in wsl. I use usbipd-win that wsl can recognise it and when I look on ```lsusb``` I can see it. but when I try to install it with instructions that I found [here](https://www.baitando.com/it/2017/12/12/install-dymo-labelwriter-on-headless-linux#:~:text=Install%20Dymo%20LabelWriter%20on%20Headless%20Linux%201%20Connection,Set%20as%20default%20printer%20...%207%20Conclusion%20) the installation failes. I suspect that it's happen becausethe latency. Since when I use the ``` sudo lpinfo -v ``` until I see the printer it takes a while. and when I try ```lpadmin -p dymo -v  usb://DYMO/LabelWriter%...``` it fails with message: ``` no matches found: usb://DYMO/LabelWriter%```.

I try to change in task manager the priority of the app but it didn't work. any other suggestions?

dorssel, 2023-08-18:
I think you are either missing a kernel driver, or you don't have enough permissions (`udev` is kind of "special" in WSL, you may need to restart it after attach).

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel