dorssel/usbipd-win #667: USB sound interface connection fails

TS-NXP, 2023-08-01:
Hi,

I'm trying to connect my device to my WSL2 instance.
I'm trying to do: 

```
usbipd wsl attach --busid 2-13
usbipd: info: Using default WSL distribution 'Ubuntu'; specify the '--distribution' option to select a different one.
usbipd: warning: USB filter 'USBPcap' is known to be incompatible with this software; 'bind --force' will be required.
```

When I run dmesg (WSL2 Ubuntu):
```
[    9.010610] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[    9.010615] vhci_hcd vhci_hcd.0: devid(131085) speed(3) speed_str(high-speed)
[    9.010654] vhci_hcd vhci_hcd.0: Device attached
[    9.369449] usb 1-1: new high-speed USB device number 2 using vhci_hcd
[    9.519452] usb 1-1: SetAddress Request (2) to port 0
[    9.555126] usb 1-1: New USB device found, idVendor=1397, idProduct=0509, bcdDevice= 1.12
[    9.555131] usb 1-1: New USB device strings: Mfr=1, Product=3, SerialNumber=0
[    9.555132] usb 1-1: Product: UMC404HD 192k
[    9.555133] usb 1-1: Manufacturer: BEHRINGER
_(Here it didn't crash yet, because I first had to do a modprobe for the sound kernel modules)_
[   44.480463] usbcore: registered new interface driver snd-usb-audio
[   45.443032] vhci_hcd: unlink->seqnum 154
[   45.443034] vhci_hcd: unlink->seqnum 155
[   45.443035] vhci_hcd: unlink->seqnum 156
[   45.443036] vhci_hcd: urb->status 0
[   45.443037] vhci_hcd: unlink->seqnum 154
[   45.443037] vhci_hcd: unlink->seqnum 155
[   45.443038] vhci_hcd: unlink->seqnum 157
[   45.443038] vhci_hcd: urb->status 0
[   45.443220] vhci_hcd: cannot find a urb of seqnum 144 max seqnum 177
[   45.443667] vhci_hcd: stop threads
[   45.443669] vhci_hcd: release socket
[   45.443674] vhci_hcd: disconnect device
[   45.443687] usb 1-1: USB disconnect, device number 2
[   45.459381] usb 1-1: 1:0: usb_set_interface failed (-19)
```

lsmod(WSL2 Ubuntu):

```
lsmod
Module                  Size  Used by
snd_usb_audio         290816  0
snd_hwdep              16384  1 snd_usb_audio
snd_usbmidi_lib        36864  1 snd_usb_audio
snd_rawmidi            36864  1 snd_usbmidi_lib
snd_pcm               118784  1 snd_usb_audio
snd_timer              40960  1 snd_pcm
snd                    86016  6 snd_hwdep,snd_usb_audio,snd_usbmidi_lib,snd_timer,snd_pcm,snd_rawmidi
soundcore              16384  1 snd
```

And then I cannot use the device. This is very strange because it was working before and then suddenly stopped working while trying to get aplay to work.

This could maybe be a bug in usbipd, but I might need some help to pin that down.

dorssel, 2023-09-03:
> [   45.443220] vhci_hcd: cannot find a urb of seqnum 144 max seqnum 177

This is (hopefully) fixed with #684 . Could you try the installer from that build (https://github.com/dorssel/usbipd-win/actions/runs/6061183362?pr=684)?

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel