dorssel/usbipd-win #669: Win 11 docker-desktop and USBIPD

doczong, 2023-08-05:
Hi All,

This isn't an issue per se with the tool, but I am at a loss and hoping someone can point me in the correct direction.

-WSL2 installed, with docker-desktop sitting on top in Windows 11.
-I have two wsl instances, docker-desktop and docker-desktop-data.
-I do not have any other linux kernel installed.
-I can't install the client tools into docker-desktop as apt-get doesn't exist.  And when I try and run the USB mapper from windows, I obviously get the "client tools not installed"

What am I missing?

If I install an additional instance (Ubuntin LTS for instance), I know I can install the client tools into it, but does that install them into the docker-desktop as well?  I'm having a senior moment as I can't wrap my head around this.

Looking to get a Sonoff Zigbee controller into Home Assistant, which is a standard docker container (homeassistant/home-assistant).

Thanks in advance.

dorssel, 2023-08-21:
All WSL instances use the same kernel. It does not matter which WSL instance is used to attach the device, it will be available in all instances. I suggest you install a dedicated Ubuntu 22.04 distro, with the client tools, just to attach the device. Then it will be available in docker as well (without requiring the client tools). Use the `usbipd wsl attach --distribution` option to select the Ubuntu distro to use for the attach. After attaching, you can shutdown the Ubuntu distro again. Once the device is attached to the WSL kernel, the dedicated client distro is then no longer needed.

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel