dorssel/usbipd-win #673: Yubikey 5 suddenly gives Device in error state message

timcooijmans, 2023-08-10:
Hi!

I was using a Yubikey 5 to have access to it SSH/GPG in a WSL2 Ubuntu instance. This worked fine for some time, but suddenly it stopped working. 

What I do:
` > usbipd.exe wsl attach -i 1050:0406` (from a normal user Powershell)
message given:
```powershell
> usbipd.exe wsl attach -i 1050:0406
usbipd: info: Device with hardware-id '1050:0406' found at busid '13-1'.
usbip: error: Attach Request for 13-1 failed - Device in error state

usbipd: error: Failed to attach device with busid '13-1'.
```

The Yubikey indeed seems to error out, it's blinking fast.

Error message in the logs:
```
Category: Usbipd.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.FileNotFoundException: Unable to find the specified file.
   at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 308
   at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
   at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
   at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 276
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 276
   at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 74
```

timcooijmans, 2023-08-10:
Update: Everything works again after a reboot

wuqijian-site, 2023-08-10:
"Unlike you, I encountered an error: Unhandled exception. System.AggregateException: One or more errors occurred. (Unrecognized line format: '{'.)".

closed 2023-08-11 by dorssel