dorssel/usbipd-win #679: Listening on port diffrent from the default?

tramora, 2023-08-17:
Hi @dorssel,
Thanx for the project.

Is there any possibility to add a **"listening port" option** when starting the usbipd server ?

I'd like to be able to customize the listening port different from the default one (3240).

I've seen the FW rule allowing the incoming flow on this port but the client still cannot connect. I suspect a filtering for this port in our organization.

dorssel, 2023-08-17:
Currently, you cannot change the listening port. I am also reluctant to make it a "standard" option (for various reasons).

It is possible (and quite easy) to add it as an option for "debugging purposes", i.e., not running as a service but from the command line. I will do that for you.

However, I think this is not the issue. If there is filtering happening, it is probably because of a 3rd party local firewall on your machine (maybe it is enforced through group policy?). Some 3rd party products honor the Windows Firewall settings, but others require specific configuration. Listening on any other port will not change that. If you don't have the rights to allow ports on such products I suggest you use SSH with reverse port forwarding; see https://github.com/dorssel/usbipd-win/issues/335#issuecomment-1221389478.

dorssel, 2023-08-17:
You can try #680. The installer is here: https://github.com/dorssel/usbipd-win/actions/runs/5892442676?pr=680.

Run on console (see https://github.com/dorssel/usbipd-win/wiki/Troubleshooting) and add appropriate options, for example:

```
usbipd server Logging:LogLevel:Default=Trace usbipd:Port=12345
```

Note that this will not work with the ```wsl``` sub-commands; you will have to run ```usbip attach``` on the client manually *with* the option to set a different port, for example:

```
sudo usbip -t 12345 attach -r <host-ip-address> -b <busid>
```

tramora, 2023-08-18:
This version works great ...
```
C:\Program Files\usbipd-win>usbipd server Logging:LogLevel:Default=Trace usbipd:Port=12345
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      usbipd:Port = 12345
dbug: Usbipd.Server[1000]
      3.1.1-PullRequest0680.16+Branch.pull-680-merge.Sha.ab7a29cea70a309fa4e54f8d0d17d2bc83549ca7
```

... unfortunatly it does not fix yet my initial issue. I'm trying to figure out the root cause.

dorssel, 2023-08-18:
> ... unfortunatly it does not fix yet my initial issue. I'm trying to figure out the root cause.

That's what I suspected. Are you sure you are not running any 3rd party security software?

tramora, 2023-08-18:
there are many on this windows host : trellix mcafee, cisco stuff... You are right, one of them must drop the connection before it reaches the (passing) usbipd fw rule

closed 2023-08-18 by tramora