dorssel/usbipd-win #699: Issue with attaching USB3V camera and Aravis and libusb drivers

Dvorkam, 2023-09-13:
using following commands I attach the device
```
PS C:\Users\michal.dvorak> usbipd bind --force -b 2-3
PS C:\Users\michal.dvorak> usbipd wsl attach -b 2-3
```

(If I don't use the first one following warning is given)
`usbipd: warning: Unknown USB filter 'tdevflt' may be incompatible with this software; 'bind --force' may be required.`

using lsusb the device is now visible 
`Bus 001 Device 003: ID 2978:d00f USB3V Camera`

I installed the aravis-tools
`sudo apt install aravis-tools`

when I run 
`sudo aravis-viewer `
the camera is shown but cannot be run and settings don't show

`sudo arv-tool-0.8`
returns 
`Company-1275-34222-9999999 (USB3)`
`sudo arv-camera-test-0.8 -d all:1`
returns
```
Looking for the first available camera
[11:12:13.734] 🆆 device> [UvDevice::read_memory] Try 1/5: command sending error: LIBUSB_ERROR_TIMEOUT
[11:12:13.768] 🆆 device> [UvDevice::read_memory] Try 2/5: command sending error: LIBUSB_ERROR_TIMEOUT
[11:12:13.801] 🆆 device> [UvDevice::read_memory] Try 3/5: command sending error: LIBUSB_ERROR_TIMEOUT
[11:12:13.835] 🆆 device> [UvDevice::read_memory] Try 4/5: command sending error: LIBUSB_ERROR_TIMEOUT
[11:12:13.868] 🆆 device> [UvDevice::read_memory] Try 5/5: command sending error: LIBUSB_ERROR_TIMEOUT
[11:12:13.868] 🆆 device> [UvDevice::_bootstrap] Error during memory read
No camera found: Failed to bootstrap USB device '(null)-(null)-(null)-1275-34222'
```
with debug lvl 3 I get number of following error messages
```
[11:14:56.105] 🅳 cp> flags        = 0x4000
                     command      = read-memory-cmd
                     size         = 12
                     id           = 65305
                     address      = 0x0000000000000004
                     size         =         64 (0x00000040)
                     00000000 55 33 56 43 00 40 00 08 0c 00 19 ff 04 00 00 00  U3VC.@..........
                     00000010 00 00 00 00 00 00 40 00                          ......@.
[11:14:56.138] 🆆 device> [UvDevice::read_memory] Try 5/5: command sending error: LIBUSB_ERROR_TIMEOUT
[11:14:56.138] 🆆 device> [UvDevice::_bootstrap] Error during memory read
```
If not using WSL, the same process is enough to retrieve images from camera.

Any idea what could be the issue?

dorssel, 2023-09-13:
Which version of `usbipd-win` are you using?
Can you run the latest with full logging + capture as explained in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting?

Dvorkam, 2023-09-13:
I was using the 3.1.0 but seeing that few days ago a 3.2.0 was released I updated and got the same result.
I reran the minimum workflow (attach, run sudo arv-camera-test-0.8 -d all:1, detach) while logging
[usb3v.zip](https://github.com/dorssel/usbipd-win/files/12598098/usb3v.zip)

dorssel, 2023-09-13:
The device repeatedly does not respond in time (timeout) when writing (commands?) to endpoint 1. This is right after configuration. The timeout appears to be 32 ms (determined by the client side software). The earlier successful configuration URBs complete within 1 ms, so I have no idea why it times out for these.