dorssel/usbipd-win #700: using  usbipd  with my own device BSOD

skunx99, 2023-09-14:
Hi all, never used  usbipd before today, so sorry for anything newb like...

I make my own usb adapters and want to work with it over WSL, after getting everything installed and trying to attach, the system make a buzzing notice and I see something about "maybe incompatible" flash up then the computer BSOD's

This is a HID gamepad device, no drives but the defaults ones with windows and linux are used. It does work on linux and on windows on its own. 

Guessing there are logs somewhere I can look at? Just not sure how to troubleshoot this one, the BSOD is a bit damaging at times. So it makes it hard to play with things.

Let me know what info I can get that would be of more help.

dorssel, 2023-09-14:
What version of `usbipd-win` are you using?

What is the BSOD stop code, and which module/driver does it list as the source of the crash?

The warning is probably about USB filter drivers. You'll get the same for `usbipd list`, without the crash, so you can read it.

skunx99, 2023-09-14:
yes the list does show it, the one it is complaining about is software not a driver.  Is there a way to tell it to be blind to it permently?

"
usbipd: warning: Unknown USB filter 'USBlyzer' may be incompatible with this software; 'bind --force' may be required.
 "

that device I wat to use is not being used by USBlyzer atm.

dorssel, 2023-09-14:
> Is there a way to tell it to be blind to it permently?

Well, yes, that is what it is telling you to do. Try `usbipd bind --help` for further help.

dorssel, 2023-09-14:
Yes, you'll need to add the required drivers to the kernel. That's a WSL issue.
Is your problem resolved now?

skunx99, 2023-09-14:
yes thx.

closed 2023-09-14 by dorssel