dorssel/usbipd-win #702: Issue with attaching after restart

dimitrijez, 2023-09-14:
Hi! I'm running Home Assistant on a docker container using WSL2 on an old laptop. The problem I have is that laptop tend to restart itself occasionally and after the restart my zigbee coordinator got detached and after that I have problem to attach it again. I'm getting an error message that usbip is not installed correctly. 

So, is there a solution in terms of keeping my coordinator attached after the restart?
Thanks

P.S. I tried everything to prevent the laptop from restarting, but nothing helped.

dorssel, 2023-09-14:
That sounds like a hardware problem (of the laptop), I'm afraid. Try `sfc /scannow` and do a repair install of `usbipd-win`.

dimitrijez, 2023-09-14:
I tried to update-alternatives in WSL, it didn't update (I already have the latest release) but that seemed to fix the problem.

Is it possible for coordinator to stay attached after the restart somehow?

dorssel, 2023-09-14:
You will have to script that yourself. Just add a startup script to Windows.

closed 2023-09-26 by dorssel