dorssel/usbipd-win #703: usbipd attach --buss-id <x> -a cannot find device if the vidpid of the device changes during use

cyborgdennett, 2023-09-14:
Hi, I am using an Coral Dev Board Micro, I use WSL2:ubuntu22(standard installation)

During the flashing process of this board, the usb vidpid-address changes.

I was using `usbipd attach --bus-id 1-2 -a` to keep the usb port alive, but when the board changes vidpid, it cannot be reached by the usbipd service, and the flashing fails. 

```ps
PS C:\WINDOWS\system32> usbipd wsl attach --busid 1-2 -a
usbipd: info: Using default WSL distribution 'Ubuntu-22.04'; specify the '--distribution' option to select a different one.
usbipd: warning: USB filter 'USBPcap' is known to be incompatible with this software; 'bind --force' will be required.
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.
Attached
Detached
Attached
Detached
usbip: error: Attach Request for 1-2 failed - Device busy (exported)
Attached
```
After the usbip error, the flash stops and reverts to old vidpid. 

Is there any know way to for example allow two hardware-id's

Thanks!

dorssel, 2023-09-14:
Sure, just bind the other VID:PID as well, and have an `--auto-attach` script running. Look through older issues/discussions; this has been solved before.

dorssel, 2023-10-20:
Stale

closed 2023-10-20 by dorssel