dorssel/usbipd-win #712: Cannot attach Yubikey - Device in error state

matrixes, 2023-09-15:
I have an issue similar to the one mentioned in https://github.com/dorssel/usbipd-win/issues/656 but I can unfortunately not get my YubiKey to attach inside WSL after having updated usbipd. When executing `usbipd wsl attach --busid 7-1`, the command is stuck for about 15 seconds, after which the YubiKey is lost from the list of devices that usbipd can see.

# Procedure

1. Start WSL
2. Start Administrator PowerShell
3. Check usbipd list, the yubikey is listed
4. Attach yubikey to wsl, command gets stuck
5. Check usbipd list, yubikey is not listed any more

I have also tried to bind the yubikey, but that makes no difference in terms of behaviour.

```
> usbipd --version
7.2.0+38.Branch.master.Sha.7ad7b9904ac0db8d759e3a53a22b1126da01b2be

> usbipd list
...
7-1    1050:0407  USB Input Device, Microsoft Usbccid Smartcard Reader (WUDF)   Not shared
...

> usbipd wsl attach --busid 7-1
usbipd: info: Using default WSL distribution 'Ubuntu'; specify the '--distribution' option to select a different one.
usbip: error: Attach Request for 7-1 failed - Device in error state

usbipd: error: Failed to attach device with busid '7-1'.

> usbipd server Logging:LogLevel:Default=Trace
...
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 172.22.155.51
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.22.155.51
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 309
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 167
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 271
         at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
dbug: Usbipd.Server[1000]
      connection close: Unable to find the specified file.
dbug: Usbipd.Server[1000]
      connection closed
```

dorssel, 2023-09-16:
What version did you upgrade from? If the driver version changed, a reboot is probably required.

matrixes, 2023-09-17:
I unfotunately don't remember what version I had before the upgrade. I'm sure it was maybe half a year ago since I did the last update/reinstall.

Anyways, the issue resolved itself after a reboot. Thanks!

closed 2023-09-17 by matrixes