dorssel/usbipd-win #714: Support for WSL2 v 2.0 in Mirrored mode

Rybasum, 2023-09-22:
The new version of WSL2 (v2.0) offers Mirrored networking mode (https://devblogs.microsoft.com/commandline/windows-subsystem-for-linux-september-2023-update). In this mode `usbipd` can't attach:

```
usbipd wsl attach -i=0403:6001
usbipd: info: Device with hardware-id '0403:6001' found at busid '4-1'.
usbipd: error: The selected WSL distribution cannot be reached via the WSL virtual switch; try restarting the WSL distribution.
```
Please, add support for this mode :-)

dorssel, 2023-09-23:
Thanks for the report!

@benmcmorran 
I'll see if I can get this mode auto-detected and supported. Can you review the result? Or any pointers for a head start?

benmcmorran, 2023-09-27:
I'm not currently working on WSL, but I can try to loop in a Microsoft engineer that is.

benmcmorran, 2023-09-27:
Sounds like reading `.wslconfig` is the only way to detect mirrored mode for now, but the team is planning an improved detection mechanism for a future WSL release.

craigloewen-msft, 2023-09-27:
@dorssel we are looking into a robust way to detect this and will share it as soon as it's available! :)

craigloewen-msft, 2023-10-05:
@dorssel we added a binary called `wslinfo` into WSL that will be able to tell you the networking mode. Are you able to use this to determine if a user is running in mirrored mode and adjust accordingly?

https://github.com/microsoft/WSL/releases/tag/2.0.4

Rybasum, 2023-10-11:
```
~$ wslinfo --networking-mode
mirrored
```
It seems to work!

dorssel, 2023-10-11:
> It seems to work!

Yes, I found the same. However ... `usbipd` needs to know the networking mode on the Windows (host) side. I have not yet decided if I want to run a remote (WSL) command (which may not be available...), just to get the information that is required on the Windows side, or instead parsing the option in `.wslconfig`.

In fact, I believe the whole WSL enumeration code needs a refactor. It is rather old, some stuff is no longer required. It was originally designed to determine the 'per-distro' IP address, which actually never worked, as the IP address is the same for all distros anyway, irrespective of the networking mode. In other words: the design doesn't really fit the purpose. Rewrite is incoming, but not very soon...

Rybasum, 2023-10-11:
Is the following helpful:
```
PS C:\Users\marci> wsl -e wslinfo --networking-mode
mirrored
```

dorssel, 2023-10-11:
Not really, sorry. This is exactly what I was referring to. This is running a remote command, on the default distro, which may not even be running (or may not even be WSL 2).

13xforever, 2023-10-17:
Reinstalling the driver doesn't help me.
When trying to attach from linux side, I can list the device fine, but when trying to attach, it fails with device error
```
root@ls:~# usbip list -r 172.28.32.1
Exportable USB devices
======================
 - 172.28.32.1
        6-2: NVIDIA Corp. : Switch [Tegra Erista] recovery mode (0955:7321)
           : USB\VID_0955&PID_7321\7&5BA284F&0&2
           : (Defined at Interface level) (00/00/00)
           :  0 - Vendor Specific Class / Vendor Specific Subclass / Vendor Specific Protocol (ff/ff/ff)

root@ls:~# usbip --debug attach -r 172.28.32.1 --busid=6-2
usbip: debug: usbip.c:129:[run_command] running command: `attach'
usbip: debug: usbip_network.c:199:[usbip_net_recv_op_common] request failed at peer: 3
usbip: error: Attach Request for 6-2 failed - Device in error state
```

on windows side I have this error in system log:
```
Category: Usbipd.ConnectedClient
EventId: 3

An exception occurred while communicating with the client:

Exception: 
System.IO.FileNotFoundException: Unable to find the specified file.
   at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 309
   at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
   at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
   at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 167
   at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 271
   at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
```

dorssel, 2023-12-06:
4.0.0 has been released, which supports mirrored networking mode.

closed 2023-12-06 by dorssel