dorssel/usbipd-win #717: BSODs keep happenning after 3.1.0

xnlcasad, 2023-09-25:
I've been following the issues raised before about Windows crashes with usbipd. See issues https://github.com/dorssel/usbipd-win/issues/461 https://github.com/dorssel/usbipd-win/issues/410 https://github.com/dorssel/usbipd-win/issues/248

It appears that https://github.com/dorssel/usbipd-win/commit/60d205dc36ac204597d5db4ef13034b6bf4c04eb commit and the 3.1.0 release was supposed to fix them.

We've been experiencing a _different_ crash from that one, and we've seen in several Windows PCs. We are using usbipd 3.1.0

The crash happens at random times, we don't really know if there is a specific scenario that triggers it.

@dorssel Any ideas on how to find the root cause here ?

Below is the crash analysis from WinDBG.

```text

************* Preparing the environment for Debugger Extensions Gallery repositories **************
   ExtensionRepository : Implicit
   UseExperimentalFeatureForNugetShare : true
   AllowNugetExeUpdate : true
   AllowNugetMSCredentialProviderInstall : true
   AllowParallelInitializationOfLocalRepositories : true

   -- Configuring repositories
      ----> Repository : LocalInstalled, Enabled: true
      ----> Repository : UserExtensions, Enabled: true

>>>>>>>>>>>>> Preparing the environment for Debugger Extensions Gallery repositories completed, duration 0.000 seconds

************* Waiting for Debugger Extensions Gallery to Initialize **************

>>>>>>>>>>>>> Waiting for Debugger Extensions Gallery to Initialize completed, duration 0.188 seconds
   ----> Repository : UserExtensions, Enabled: true, Packages count: 0
   ----> Repository : LocalInstalled, Enabled: true, Packages count: 36

Microsoft (R) Windows Debugger Version 10.0.25921.1001 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\Users\xxx\Downloads\MEMORY.DMP]
Kernel Bitmap Dump File: Kernel address space is available, User address space may not be available.

Symbol search path is: srv*
Executable search path is: 
Windows 10 Kernel Version 19041 MP (8 procs) Free x64
Product: WinNt, suite: TerminalServer SingleUserTS
Edition build lab: 19041.1.amd64fre.vb_release.191206-1406
Kernel base = 0xfffff805`43a00000 PsLoadedModuleList = 0xfffff805`4462a3d0
Debug session time: Thu Sep 21 13:21:39.229 2023 (UTC + 2:00)
System Uptime: 7 days 4:42:35.116
Loading Kernel Symbols
...............................................................
................................................................
................................................................
....................................................
Loading User Symbols
PEB is paged out (Peb.Ldr = 000000e9`6f398018).  Type ".hh dbgerr001" for details
Loading unloaded module list
..................................................
For analysis of this file, run !analyze -v
nt!KeBugCheckEx:
fffff805`43dfcc40 48894c2408      mov     qword ptr [rsp+8],rcx ss:0018:ffffa689`6f327610=0000000000000018
3: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

REFERENCE_BY_POINTER (18)
Arguments:
Arg1: 0000000000000000, Object type of the object whose reference count is being lowered
Arg2: ffff9689558029d0, Object whose reference count is being lowered
[... 226 lines cut ...]

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {3ade888f-df39-202d-9e7b-2930c63fbded}

Followup:     MachineOwner
---------

3: kd> lmvm VBoxUSBMon
Browse full module list
start             end                 module name
fffff805`617f0000 fffff805`6182a000   VBoxUSBMon   (export symbols)       VBoxUSBMon.sys
    Loaded symbol image file: VBoxUSBMon.sys
    Image path: \SystemRoot\system32\DRIVERS\VBoxUSBMon.sys
    Image name: VBoxUSBMon.sys
    Browse all global symbols  functions  data
    Timestamp:        Wed Jul 12 18:34:34 2023 (64AED61A)
    CheckSum:         0003BF48
    ImageSize:        0003A000
    Translations:     0000.04b0 0000.04e4 0409.04b0 0409.04e4
    Information from resource tables:

```

dorssel, 2023-09-26:
Thanks for the report.

This is indeed different from earlier BSOD, which happened in VBoxUSB.sys. This one is in VBoxUSBMon.sys instead.

Normally, VBoxUSBMon is dormant until either a device is plugged/unplugged or attached/detached. When you say "at random times", is there any relation with USB devices changing state? Are there any USB filter drivers installed (`usbipd-win list` would report those)?

In any case, it is an issue with the VBoxUSBMon driver, so you may want to report this to VirtualBox as well...

xnlcasad, 2023-09-26:
Report raised on their [bugtracker](https://www.virtualbox.org/ticket/21859)

mateuszdrab, 2024-05-09:
Just experienced a BSOD today on my hypervisor which I recently configured to run usbipd to share a Google Coral stick to a VM. 

I did notice the application using Coral was regularly restarting due to losing the USB device.

I did notice the error from the below issue in my logs
https://github.com/dorssel/usbipd-win/issues/423

Perhaps the constant resetting of the USB device resulted in the BSOD

dorssel, 2024-05-09:
@mateuszdrab 
Did you see which module / driver caused the BSOD? And what was the STOP code?

mateuszdrab, 2024-05-09:
@dorssel sure, here's the entire windbg analyze output.

```
Microsoft (R) Windows Debugger Version 10.0.22621.755 AMD64
Copyright (c) Microsoft Corporation. All rights reserved.

Loading Dump File [C:\temp\050924-46625-01.dmp]
Mini Kernel Dump File: Only registers and stack trace are available

Symbol search path is: srv*
Executable search path is: 
Windows 10 Kernel Version 20348 MP (88 procs) Free x64
Product: Server, suite: TerminalServer DataCenter SingleUserTS
Machine Name:
Kernel base = 0xfffff803`80200000 PsLoadedModuleList = 0xfffff803`80e33de0
Debug session time: Thu May  9 10:12:45.731 2024 (UTC + 1:00)
System Uptime: 10 days 18:45:29.032
Loading Kernel Symbols
...............................................................
................................................................
................................................................
........
Loading User Symbols
Loading unloaded module list
..................................................
For analysis of this file, run !analyze -v
0: kd> !analyze -v
*******************************************************************************
*                                                                             *
*                        Bugcheck Analysis                                    *
*                                                                             *
*******************************************************************************

DPC_WATCHDOG_VIOLATION (133)
The DPC watchdog detected a prolonged run time at an IRQL of DISPATCH_LEVEL
or above.
Arguments:
Arg1: 0000000000000001, The system cumulatively spent an extended period of time at
	DISPATCH_LEVEL or above. The offending component can usually be
	identified with a stack trace.
Arg2: 0000000000001e00, The watchdog period.
Arg3: fffff80380f0f328, cast to nt!DPC_WATCHDOG_GLOBAL_TRIAGE_BLOCK, which contains
	additional information regarding the cumulative timeout
Arg4: 0000000000000000

Debugging Details:
------------------

*** WARNING: Unable to verify timestamp for VBoxUSB.sys
*************************************************************************
***                                                                   ***
***                                                                   ***
***    Either you specified an unqualified symbol, or your debugger   ***
***    doesn't have full symbol information.  Unqualified symbol      ***
***    resolution is turned off by default. Please either specify a   ***
***    fully qualified symbol module!symbolname, or enable resolution ***
***    of unqualified symbols by typing ".symopt- 100". Note that     ***
***    enabling unqualified symbol resolution with network symbol     ***
***    server shares in the symbol path may cause the debugger to     ***
***    appear to hang for long periods of time when an incorrect      ***
***    symbol name is typed or the network symbol server is down.     ***
***                                                                   ***
***    For some commands to work properly, your symbol path           ***
***    must point to .pdb files that have full type information.      ***
***                                                                   ***
***    Certain .pdb files (such as the public OS symbols) do not      ***
***    contain the required information.  Contact the group that      ***
***    provided you with these symbols if you need this command to    ***
***    work.                                                          ***
***                                                                   ***
[... 84 lines cut ...]
ffffa788`150b66f8 00000000`00000000     : 00000000`00000000 00000000`00000000 00000000`00000000 00000000`00000000 : 0x2

SYMBOL_NAME:  USBXHCI!Endpoint_UcxEvtEndpointAbort+c2

MODULE_NAME: USBXHCI

IMAGE_NAME:  USBXHCI.SYS

IMAGE_VERSION:  10.0.20348.1450

STACK_COMMAND:  .cxr; .ecxr ; kb

BUCKET_ID_FUNC_OFFSET:  c2

FAILURE_BUCKET_ID:  0x133_ISR_USBXHCI!Endpoint_UcxEvtEndpointAbort

OSPLATFORM_TYPE:  x64

OSNAME:  Windows 10

FAILURE_ID_HASH:  {0c82b582-a7fa-76d8-0eac-8fe9bb0424be}

Followup:     MachineOwner
---------
```

dorssel, 2024-05-09:
That's the driver of the hub the device is connected to (`USBXHCI.SYS`). Can you find that hub in the Windows Device Manager. Is it the stock Microsoft driver, or an OEM driver?

mateuszdrab, 2024-05-10:
It looks like it is the stock driver.

[image]

dorssel, 2024-05-10:
@mateuszdrab 
Well, that's a bug in a Microsoft driver then. Better report it. Nothing I can do about that.

mateuszdrab, 2024-05-10:
I can try to reproduce it on another system I don't mind BSODing, using both stock and Intel drivers and see if it makes any difference.