dorssel/usbipd-win #721: Issue with WIX Installer or Driver

zadmoe, 2023-10-01:
Hi, I'm having a problem and I can't figureout whats causing it, 
I build the project from your source with no problem but when i try to install from the msi thats created it fails on trying to instll the drivers and rolls back, I have tried on different computers with luck. I can install the msi that i download with no problem so It must be something wrong with my environment but I have trying downloading the source and building without making any changes but it won't work.
------------------
DIFXAPP: INFO:   RETURN: DriverPackageInstallW  (0xE000024B)
DIFXAPP: ERROR: encountered while installing driver package 'C:\Program Files\usbipd-win\Drivers\VBoxUSB\VBoxUSB.inf'
DIFXAPP: ERROR: InstallDriverPackages failed with error 0xE000024B
DIFXAPP: RETURN: InstallDriverPackages() 3758096971 (0xE000024B)
CustomAction MsiInstallDrivers returned actual error code 1603 (note this may not be 100% accurate if translation happened inside sandbox)
Action ended 11:31:42: InstallExecute. Return value 3.
------------------
I would appretiate your help greatly.

dorssel, 2023-10-01:
What OS are you running?

`0xE000024B` means driver signature problems. However, the repo contains properly signed drivers. Did you also compile those yourself (you probably shouldn't...)?

The installer on the release page is the exact same as the one from the build artifacts, except that I additionally signed the MSI itself (and the dll and exe contained in the MSI). The drivers within are already signed by Oracle.

dorssel, 2023-10-01:
See https://github.com/Seeed-Studio/Signed_USB_Serial_Driver/issues/4 for a similar issue. You didn't change anything in the INF file, I hope?

zadmoe, 2023-10-02:
Hello again, I'll check these and will let you know in a few days. Thank You

zadmoe, 2023-10-07:
Hi Frans, sorry for the delay, I was away for a few days, 
I'm using windows 10, and I have tried runnig the installer on other machines and I get the same error. I am trying to make a few changes to usbipd but since I got stuck, I downloaded your source again and without making any changes( exept adding the installer to build configuration) I build your solution and i'm getting this error. 
I tried to check the cat file using signtool and Its verified.

dorssel, 2023-10-20:
Can you post your installer?

zadmoe, 2023-10-21:
[usbipd-win_0.1.0.zip](https://github.com/dorssel/usbipd-win/files/13059939/usbipd-win_0.1.0.zip)

dorssel, 2023-10-23:
Found it. It turns out that you checked out using a Linux line ending (LF) convention. The .inf files of the drivers are indeed text, but the line endings must be Windows style (CRLF). If not, the signature will be corrupted.

This will be fixed by https://github.com/dorssel/usbipd-win/pull/739. For the time being, either check out using Windows, or convert the files back to CRLF with a Linux tool like `unix2dos` (or so...).

zadmoe, 2023-10-24:
Wow, you are a genius, I have no Idea how you figured that out. 
I guess this is because I didn't use Git and I just downloaded the zip of the source from here and somehow the driver's .ini file was LF eventhought I am on windows. All good now. Thank you so much.

closed 2023-10-24 by zadmoe