dorssel/usbipd-win #723: System.IO.FileNotFoundException error when trying to attach either usb device to wsl

anthonytrouble, 2023-10-06:
Hi there

First, congrats on this great contribution!

I'm facing an issue while trying to connect **any usb device** (bus id) to wsl.

Here is the basic answer when trying to attach:
```shell
PS C:\Windows\system32> usbipd wsl attach -b 1-14
usbipd: info: Using default WSL distribution 'Ubuntu-20.04'; specify the '--distribution' option to select a different one.
usbip: error: Attach Request for 1-14 failed - Device in error state
``` 

And here is the detailed log:
```shell
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.IO.FileNotFoundException: Unable to find the specified file.
         at Usbipd.ConfigurationManager.GetVBoxDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\ConfigurationManager.cs:line 309
         at Usbipd.VBoxUsb.ClaimDeviceOnce(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 66
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.VBoxUsb.ClaimDevice(BusId) in D:\a\usbipd-win\usbipd-win\Usbipd\VBoxUsb.cs:line 87
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 167
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 271
         at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
``` 

I'm using (latest) version 3.2.0... I'm sure there must be a small detail that I missed, but tried to reboot, then uninstall / reinstall without success.

Thanks

dorssel, 2023-10-12:
This usually means that the VBoxUSB driver doesn't load properly. You could try binding with `--force` followed by a reboot.

anthonytrouble, 2023-10-16:
Thanks @dorssel for your reply.

I tried using `--force` but it was not solving the issue. However it seems like the issue was due to one of the connected USB devices not being properly powered... I controlled the AC adapter and from then on, everything was working as expected.

Thanks again for your reply and for the great work done here!

closed 2023-10-16 by anthonytrouble