dorssel/usbipd-win #731: Error building custom kernel

Joly0, 2023-10-15:
Hey, i have tried building  the custom kernel with this command "sudo make -j 8 && sudo make modules_install -j 8 && sudo make install -j 8" but i get the following output:
[image]
Any idea why the error "FAILED: load BTF from vmlinux: Unknown error -22make: *** [Makefile:1179: vmlinux] Error 255" appears here? I tried googling it, but i havent found anything useful

dorssel, 2023-10-19:
Depending on which additional drivers you enabled, you may have to install additional developer packages. The instructions on https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel are just for the default kernel configuration.

Which additional CONFIG_XXX did you add?

vinceg77, 2023-11-08:
a `sudo apt install dwarves` should resolve it, then re run the compilation