dorssel/usbipd-win #734: USB stability issues on long running connections with Eflow-VM

mamelea, 2023-10-17:
Hi,

we use this Usbipd-Win in combination with Azure Eflow-VM.  (so we use this lib indirectly to establish an passthrough for this Eflow-VM) 

**What we want to achieve:**
Passthrough of an HID-USB device, long term. (actually this should USB-Device should be available the whole uptime of the Eflow-VM, which could be months)

**What we experience:**
USB Device suddenly disappears, sometimes after mins, sometimes after hours. 'lsusb' doesnt list the device anymore.
The only workflow which does help, is it "mount" it again via "Add-EflowUSBDevices"

Could this be a problem of the Eflow-VM?
How can we debug long running passthroughs with usbipd?

Any help is gladly appreciated. Thank you.

dorssel, 2023-10-19:
First you want to figure out what is causing the disconnect. Run the server on the console (with full logging), as explained in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting. Also make sure the you get the syslog on the Linux side of things.

Once the disconnect reason is known, we'll take the next debugging step (depending on the outcome).

mamelea, 2023-11-07:
In the meantime, we have switched to another device that offers the same functionality via Ethernet. It runs quite stable. I don't think we'll be putting any more effort into the USB solution.  (as it maybe an issue of faulty hardware or faulty USB implementation)

Thank you for your help.

closed 2023-11-07 by mamelea