dorssel/usbipd-win #735: Usb device is detected by lsusb and dmesg, but not displayed

TK1987, 2023-10-18:
Hi there.

I am trying to connect a usb stick under Windows 10 22H2 (Build 19045.3570) in WSL2.
I followed the [guide](https://github.com/dorssel/usbipd-win/wiki/WSL-support) to build my own kernel.

During compilation I got the following message, if that is somehow important:
```
arch/x86/Makefile:142: CONFIG_X86_X32 enabled but no binutils support
```

After starting ubuntu `uname -r` shows that the kernel has been loaded:
```
5.15.90.1-microsoft-standard-WSL2+
```

After attaching the usb stick, `lsusb` also displays the device:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 0781:55a9 SanDisk Corp.  SanDisk 3.2Gen1
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

However, with `lsblk` it is not displayed.

```
$ dmesg | grep -i usb
[    0.093543] ACPI: bus type USB registered
[    0.093550] usbcore: registered new interface driver usbfs
[    0.093554] usbcore: registered new interface driver hub
[    0.093558] usbcore: registered new device driver usb
[    0.304803] usbcore: registered new interface driver cdc_ether
[    0.305190] usbcore: registered new interface driver cdc_ncm
[    0.305572] usbcore: registered new interface driver r8153_ecm
[    0.306824] usbcore: registered new interface driver cdc_acm
[    0.307327] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[    0.307823] usbcore: registered new interface driver ch341
[    0.308104] usbserial: USB Serial support registered for ch341-uart
[    0.308451] usbcore: registered new interface driver cp210x
[    0.308729] usbserial: USB Serial support registered for cp210x
[    0.309076] usbcore: registered new interface driver ftdi_sio
[    0.309438] usbserial: USB Serial support registered for FTDI USB Serial Device
[    0.309962] vhci_hcd vhci_hcd.0: USB/IP Virtual Host Controller
[    0.310323] vhci_hcd vhci_hcd.0: new USB bus registered, assigned bus number 1
[    0.311259] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.15
[    0.311685] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.312094] usb usb1: Product: USB/IP Virtual Host Controller
[    0.312502] usb usb1: Manufacturer: Linux 5.15.90.1-microsoft-standard-WSL2+ vhci_hcd
[    0.312914] usb usb1: SerialNumber: vhci_hcd.0
[    0.313326] hub 1-0:1.0: USB hub found
[    0.314097] vhci_hcd vhci_hcd.0: USB/IP Virtual Host Controller
[    0.314446] vhci_hcd vhci_hcd.0: new USB bus registered, assigned bus number 2
[    0.314944] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
[    0.315519] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.15
[    0.315961] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.316381] usb usb2: Product: USB/IP Virtual Host Controller
[    0.316783] usb usb2: Manufacturer: Linux 5.15.90.1-microsoft-standard-WSL2+ vhci_hcd
[    0.317193] usb usb2: SerialNumber: vhci_hcd.0
[    0.317586] hub 2-0:1.0: USB hub found
[    0.324030] usbcore: registered new interface driver usbhid
[    0.324318] usbhid: USB HID core driver
[  911.258135] usb 1-1: new high-speed USB device number 2 using vhci_hcd
[  911.408205] usb 1-1: SetAddress Request (2) to port 0
[  911.444656] usb 1-1: New USB device found, idVendor=0781, idProduct=55a9, bcdDevice= 1.00
[  911.444664] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  911.444667] usb 1-1: Product:  SanDisk 3.2Gen1
[  911.444669] usb 1-1: Manufacturer:  USB
[  911.444670] usb 1-1: SerialNumber: 01016156e64ed345e2dcec4df2dfee18093815751d4157acd91d7116dd6ccf56b22c00000000000000000000965e4532000e7300a955810737acb033```

Any suggestions?

Regards, Thomas

dorssel, 2023-10-19:
Looks like you did not enable USB storage support into your kernel. Or: you are not actually using the kernel you built yourself, but still the default. In any case: the device is attaching properly, but Linux does not load the storage driver for it, so it just sits there, idly. This is a WSL issue, for support https://github.com/microsoft/WSL/issues

TK1987, 2023-10-19:
Thanks for the reply.

> Looks like you did not enable USB storage support into your kernel.

You're right. I followed the guide exactly, but of course I should have enabled
```
Device Drivers -> USB Support -> USB Mass Storage support
```
as well.

It works now. Thank you very much.

closed 2023-10-19 by TK1987