dorssel/usbipd-win #736: usbip client not correctly installed - followed other threads, still can't get it working

sne-samal, 2023-10-20:
I have followed instructions outlined in the wiki to try to connect to a USB port in wsl, I am not using docker, and the error message that wsl usbip client not correctly installed is still coming up. I have gone through some of the potential solutions in other threads, but no difference has been made. Any help appreciated

dorssel, 2023-10-21:
What distro?

sne-samal, 2023-10-21:
WSL 2

dorssel, 2023-10-21:
That's not the distro. Which Linux installation are you using? What is the output of `wsl --list --verbose`?

sne-samal, 2023-10-21:
This is the output:
* Ubuntu 
Ubuntu-22.04
Both are running, version 2

dorssel, 2023-10-21:
> I have followed instructions outlined in the wiki

What is the output of the `update-alternatives` command from the instructions?

sne-samal, 2023-10-21:
When running this:
```
sudo apt install linux-tools-virtual hwdata
sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20
```
I get:
0 upgraded, 0 newly installed, 0 to remove and 52 not upgraded.

dorssel, 2023-10-21:
What is the output if you manually run `usbip` from within WSL?

sne-samal, 2023-10-21:
It initially returned, no such command, then I entered the installation instructions into wsl, before, i put it into a Ubuntu 22.04.2 LTS terminal and tried to connect to the USB, this now works - successfully attached to a USB device through WSL

dorssel, 2023-10-21:
So, you actually want to use "Ubuntu-22.04", but the default is "Ubuntu"?

sne-samal, 2023-10-21:
Yes

dorssel, 2023-10-21:
Either of these option:

1) always use `usbipd wsl ...` with the `---distribution` option

or 

2) change the default WSL distro with `wsl --set-default`

dorssel, 2023-12-06:
4.0.0 has been released and installing client-side tooling is no longer necessary.

closed 2023-12-06 by dorssel