dorssel/usbipd-win #740: Run away cache usage in WSL + error logging using disk space

matthew-william-lock, 2023-10-24:
Hey!

I have a bit of a strange one. I am running WSL and realized today that my RAM usage in Windows was filling up like crazy when running software that interfaces with a USB radio transceiver through usbipd-win. After some more investigation, I realized that my system logs were filling up at an incredible rate (gigs within minutes)

Attached is a snippet of the logs from /var/log/kern.log

```
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.265995] vhci_hcd: pickup_urb_and_free_priv:26: find urb seqnum 66983
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.265997] usbip_core: usbip_recv:314: enter
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.265998] usbip_core: usbip_recv:327: receiving....
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.265999] usbip-core00000000: 00
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266000] usbip_core: usbip_recv:329: received, osize 1 ret 1 size 0 total 1
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266002] usb 1-1:        devnum(2) devpath(1) usb speed(full-speed)
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266004] usbip_core: usbip_dump_usb_device:94: tt hub ttport 1
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266005] usb 1-1:
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266006] usbip_core: usbip_dump_usb_device:98:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266007] usbip_core: usbip_dump_usb_device:98:   1
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266007] usbip_core: usbip_dump_usb_device:98:   2
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266008] usbip_core: usbip_dump_usb_device:98:   3
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266009] usbip_core: usbip_dump_usb_device:98:   4
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266009] usbip_core: usbip_dump_usb_device:98:   5
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266010] usbip_core: usbip_dump_usb_device:98:   6
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266011] usbip_core: usbip_dump_usb_device:98:   7
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266011] usbip_core: usbip_dump_usb_device:98:   8
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266012] usbip_core: usbip_dump_usb_device:98:   9
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266013] usbip_core: usbip_dump_usb_device:98:  10
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266014] usbip_core: usbip_dump_usb_device:98:  11
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266014] usbip_core: usbip_dump_usb_device:98:  12
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266015] usbip_core: usbip_dump_usb_device:98:  13
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266016] usbip_core: usbip_dump_usb_device:98:  14
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266016] usbip_core: usbip_dump_usb_device:98:  15
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266017] usbip_core: usbip_dump_usb_device:99:
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266018] usb 1-1:        toggle0(IN) :
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266019] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266020] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266020] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266021] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266022] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266023] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266023] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266024] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266025] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266025] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266026] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266027] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266027] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266028] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266029] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266030] usbip_core: usbip_dump_usb_device:103:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266030] usbip_core: usbip_dump_usb_device:104:
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266031] usb 1-1:        toggle1(OUT):
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266032] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266032] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266033] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266034] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266034] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266035] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266036] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266036] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266037] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266038] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266039] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266039] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266040] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266041] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266041] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266042] usbip_core: usbip_dump_usb_device:108:   0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266043] usbip_core: usbip_dump_usb_device:109:
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266043] usb 1-1:        epmaxp_in   :
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266044] usbip_core: usbip_dump_usb_device:114:  64
[... 19 lines cut ...]
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266063] usb 1-1:    error_count           :0
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266063] vhci_hcd: vhci_recv_ret_submit:101: now giveback urb 66983
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266097] vhci_hcd: vhci_recv_ret_submit:109: Leave
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266100] vhci_hcd: vhci_rx_pdu:203: Enter
Oct 17 13:02:33 LAPTOP-1532NB09 kernel: [  191.266101] usbip_core: usbip_recv:314: enter
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094186] cfclient[915]: segfault at 55cba10fe5d0 ip 000055cba10fe5d0 sp 00007ffee79cc818 error 15
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094192] Code: 00 00 0d 00 00 00 00 00 00 00 63 6c 69 63 6b 65 64 28 62 6f 6f 6c 29 00 00 00 e0 25 37 24 d9 7f 00 00 61 00 00 00 00 00 00 00 <f0> 8c 0a a1 cb 55 00 00 00 8d 0a a1 cb 55 00 00 01 00 00 00 00 00
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094196] potentially unexpected fatal signal 11.
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094196] CPU: 0 PID: 915 Comm: cfclient Not tainted 5.15.90.1-microsoft-standard-WSL2+ bitcraze/crazyflie-clients-python#2
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094198] RIP: 0033:0x55cba10fe5d0
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094201] Code: 00 00 0d 00 00 00 00 00 00 00 63 6c 69 63 6b 65 64 28 62 6f 6f 6c 29 00 00 00 e0 25 37 24 d9 7f 00 00 61 00 00 00 00 00 00 00 <f0> 8c 0a a1 cb 55 00 00 00 8d 0a a1 cb 55 00 00 01 00 00 00 00 00
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094202] RSP: 002b:00007ffee79cc818 EFLAGS: 00010246
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094203] RAX: 000055cba1de09f0 RBX: 00007fd8cd6be160 RCX: 00007fd91908e640
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094204] RDX: 00007fd9191fef10 RSI: 00007fd91908e650 RDI: 000055cba27f1430
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094204] RBP: 00007fd91908e650 R08: 000000000000000b R09: 000055cba082f590
[... cut ...] el: [  399.094205] R10: 000055cba084bdc0 R11: 000055cb9f116380 R12: 00007fd8cd6be160
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094206] R13: 000055cba1872020 R14: 00007fd91d2d6640 R15: 000055cba082f590
Oct 17 13:06:01 LAPTOP-1532NB09 kernel: [  399.094207] FS:  00007fd9248a11c0 GS:  0000000000000000
```

Is there some way I can stop this logging or see what the actual issue is?

See https://github.com/orgs/bitcraze/discussions/981 for the original issue,

Any help is sincerely appreciated! 😄

matthew-william-lock, 2023-10-24:
For more context, the USB device is working fine :) I just need to stop this crazy amount of logging as it eats my system resources...

matthew-william-lock, 2023-10-24:
I am able to stop the logging from filling up my file system by doing this https://www.thinkpenguin.com/gnu-linux/how-disable-logging-varlogkernlog, but my cache is still filling up so I imagine it is still trying to log

dorssel, 2023-10-24:
Did you build the kernel yourself? Note the remark about debug logging in Linux in https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel. Better turn that off...

If there is excessive logging to syslog without that debug option, then there may be something else wrong.

matthew-william-lock, 2023-10-24:
I have not built the kernel myself. I simply followed the [normal installation](https://learn.microsoft.com/en-us/windows/wsl/connect-usb#install-the-usbipd-win-project) instructions. Is logging turned off in this installation?

dorssel, 2023-10-24:
It should be turned off in the default WSL kernel build. What version are you running? (`wsl --info`)

matthew-william-lock, 2023-10-24:
```wsl --info``` does not seem to work, but ```wsl -v``` returns the following 

```
WSL version: 1.2.5.0
Kernel version: 5.15.90.1
WSLg version: 1.0.51
MSRDC version: 1.2.3770
Direct3D version: 1.608.2-61064218
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.22621.2428
```

dorssel, 2023-10-25:
Are you absolutely sure you are running the stock Windows WSL kernel? Nothing in `<HOME>/.wslconfig`?

matthew-william-lock, 2023-10-25:
```<HOME>/.wslconfig``` shows:

```
[wsl2]
kernel=c:\\users\\mtthe\\usbip-bzImage
```

with a ```usbip-bzImage``` file there...

Is this not the stock?

matthew-william-lock, 2023-10-25:
@dorssel I might have built the kernel ages ago and not realized that this config file was kept despite a Windows reset... I have removed the config file and all is working as it should be! Thank you and sorry for the trouble 😅

closed 2023-10-25 by matthew-william-lock