dorssel/usbipd-win #742: usbipd requires admin privileges

tpitman, 2023-10-25:
I have install the app store version of wsl 2.

I would like to attach usb devices to that wsl installation.

I installed usbipd and have the wsl shell open and running.

I open another cmd prompt and try to run usbipd wsl attach --busid 10-3. I get an error that this requires admin privileges.

I tried openning a cmd prompt as administrator and then when I run the command it tells me I don't have any linux distributions installed. I can only assume that is because the one I did install is installed for the logged in user and administrator command prompts use a different wsl and therefore don't have one installed.

I need the wsl to be running for the logged in user and not the administrator, so I can't just install a wsl linux distribution while in an admin cmd prompt.

I tried opening a regular cmd prompt and using the runas command to temporarily elevate that window to run the usbipd attach command. When I do this a command prompt opens and quickly closes. I can't see any output in that window, but lsusb in the linux distribution still doesn't show any devices, so it didn't work.

What do I do?

dorssel, 2023-10-25:
What is the output of the `usbipd wsl attach` command when you run it as a normal user?

tpitman, 2023-10-26:
Nevermind. I got it working. The last step was simply doing `sudo apt-get update` and `sudo apt-get install usbip`.

I thought I had tried that before and it couldn't find usbip as a package. Not sure what fixed that.

Now when I do the attach command it works. Not sure if the device works yet, but at least this part is working.

closed 2023-10-26 by tpitman