dorssel/usbipd-win #745: RNDIS Support for USB Network Devices

tpitman, 2023-10-27:
I have a device based on Beaglebone Black. We use the bootloader over RNDIS and USB to program these devices.

When connected to a Debian linux computer that we have set up, our device shows up as a usb0 network device over RNDIS.

We would like to replace the linux computer with a computer running Windows 10 because we already have that as well for other parts of the programming / setup process.

I installed WSL and usbip. Those seem to be working. I also followed your instructions on generating a new kernel, not for the usbip support because that is already in the latest kernel, but rather to turn on the RNDIS support.

I can't seem to get our device to enumerate as usb0.

I turned on all of the RNDIS settings I could find using menuconfig. I built the kernel and the bzImage and copied that over to the host Windows and made the .wslconfig file as instructed.

I can see my device in lsusb. I cannot see the usb0 using ifconfig.

I could use some help to figure out what is still missing.

dorssel, 2023-10-27:
You need to recompile the kernel with the correct USB driver for your USB network device. Microsoft only enables a handful of serial-to-USB device drivers in their stock WSL kernel. Currently, Linux attaches your USB device (and it shows up in `lsusb`), but then it does not have an actual driver for it (so it never actually gets used by the Linux side). In other words: you need a specific USB driver compiled into the kernel to actually make the USB device show up as a network device.

Without the VID:PID of the device, I cannot tell which driver you would need.

tpitman, 2023-10-27:
Thank you for the quick reply. This is supposed to be a pretty common and standard driver. It is the Beaglebone Black device.

Sorry I didn't include the VID/PID. It is: 0451:6141. It is the Texas Instruments AM335x.

Would you be able to tell me where I can turn on compiling that driver into the kernel using menuconfig or some other method?

dorssel, 2023-10-27:
Does this maybe have something to do with it? https://news.itsfoss.com/linux-disable-microsoft-rndis/

dorssel, 2023-10-27:
I guess you need to enable this one: https://developer.nordicsemi.com/nRF_Connect_SDK/doc/v1.5-branch/kconfig/CONFIG_USB_DEVICE_NETWORK_RNDIS.html

tpitman, 2023-10-27:
That is possible.

I cannot find CONFIG_USB_DEVICE_NETWORK_RNDIS in menuconfig anywhere. That webpage seems to be specific to Nordic and their zepher stuff. Thanks for the tip, though.

Any other thoughts?

dorssel, 2023-10-28:
I found this post:

https://www.linuxquestions.org/questions/linux-networking-3/using-a-detected-usb-rndis-gadget-as-network-adapter-4175698461/#post6270910

dorssel, 2024-02-04:
stale

closed 2024-02-04 by dorssel