dorssel/usbipd-win #747: FR: Share a USB HUB

lelemka0, 2023-11-01:
Is it possible to share an entire usb hub?
When the client attaches to a hub, all devices on it will be connected to the client. This can also be a way to implement auto-attach and works for any client (wsl/vm/other host).

dorssel, 2023-11-02:
The simple answer is: no you cannot share a hub (or even a port). This is not in the USBIP protocol, and none of the drivers (Windows or Linux) support it.

The more complex answer is: you could run a service/script on the Windows side to auto-bind anything newly connected. But you then *also* have to run a service/script on the client that periodically polls the host if any such newly detected devices are available. And then the client needs to attach those. The reason is that the USBIP protocol does not support "pushing" a device attach, it needs to be requested client side. WSL is special in the sense that the host has a way to run remote commands, but this is not possible generically for any other clients. As such, this is outside the scope of `usbipd-win`, which is a generic solution for any USBIP clients, not just WSL.

TLDR; This requires additional software, much like the GUI options advertised in the README. It is outside the scope of this project.

lelemka0, 2023-11-02:
thanks for clarifying my doubts

closed 2023-11-02 by lelemka0