dorssel/usbipd-win #748: Unable to install at Windows 10 LTSC

dimangelid, 2023-11-04:
Hello,

I am trying to install usbipd-win_3.2.0.msi at my laptop. I have run the installation with administrator priviledges.

I use Windows 10 LTSC 64bit , version 1809, OS Build 17763.4851

The error after attempting to install the above msi file, is:
**Unsuccessful**
**The operation ended prematurely because of an error. Your computer was not modified**

I don't have Windows Subsystem for Linux at this laptop. My goal is to share a USB device with a Linux virtual machine, which runs inside Virtualbox.

I have run the installation via CMD ( msiexec /i usbipd-win_3.2.0.msi /L*vx usbipd-win.log ) in order to generate the attached log file
[usbipd-win.log](https://github.com/dorssel/usbipd-win/files/13255517/usbipd-win.log) . I run CMD as an administrator.

My Virtualbox version is 7.0.97 r159680 (Qt6.3.0) and has the latest extension pack that is compatible with the Virtualbox version.

Does anyone has an idea why the installation fails?

dorssel, 2023-11-04:
`usbipd-win` uses the VirtualBox drivers. In the past, I've noticed that it isn't playing nice with a full installation of VirtualBox. The best you can do is make sure that the VirtualBox version matches the version of the drivers `usbipd-win` comes with, currently 7.0.8. And the order of installation matters: probably VirtualBox first, and `usbipd-win` last. Your log shows driver installation problems, probably because the versions don't match.

On the other hand: (this may seem a bit odd from the author of `usbipd-win`) Why are you using `usbipd-win` at all? VirtualBox has very nice USB passthrough support built-in already. Unless you want to *also* share your devices with other systems besides VirtualBox guests, then there is no need for `usbipd-win`.

dimangelid, 2023-11-04:
Hello dorssel,

I have tried again with the below Virtualbox versions:

VirtualBox-7.0.8-156879-Win and extension pack Oracle_VM_VirtualBox_Extension_Pack-7.0.8

VirtualBox-7.0.10-158379-Win and extension pack Oracle_VM_VirtualBox_Extension_Pack-7.0.10

Again usbipd-win installation failed. I have gathered logs for each installation attempt:

[usbipd-win_vbox708.log](https://github.com/dorssel/usbipd-win/files/13257872/usbipd-win_vbox708.log)
[usbipd-win_vbox7010.log](https://github.com/dorssel/usbipd-win/files/13257873/usbipd-win_vbox7010.log)

As for the usbipd-win usage reason, i don't want to use the USB passthrough feature of Virtualbox. I want this USB device to be shared both with the host and the virtual machine. And with the USB passthrough of Virtualbox, it will only be usable from the virtual machine.

dorssel, 2023-11-04:
Both logs show driver install failure.

But ... `usbipd-win` works *exactly* the same as VirtualBox. You will have to attach a device to the guest, after which it is not usable by the host. And you detach again, after which it is available to the host again. You cannot use the device on the host and guest simultaneously.

dimangelid, 2023-11-05:
Thanks @dorssel , i was not aware that the device will be unavailable to the host.

closed 2023-12-18 by dorssel