dorssel/usbipd-win #751: USB serial converter only attaches for 5-10 seconds before unattaching.

Tapir28, 2023-11-06:
I have a usb to serial converter which has been bound from powershell and attached with `usbip attach -r localhost --busid=xxx` this works and the device now shows up in wsl with `lsusb`.

After several seconds the device will disconnect, reappear in windows device manager, and no longer shows up in wsl `lsusb`.

This happens every time I attach the device. A different serial converter which is working on a seperate machine also has the same results. Changing which usb port is used also has the same results. Anywhere from 5-15 seconds and it unattaches itself.

WSL 2.0.7 set up in mirrored mode
usbipd 3.2.0
usbip 2.0

dorssel, 2023-11-06:
Can you get all diagnostics (full logging + capture) as per https://github.com/dorssel/usbipd-win/wiki/Troubleshooting?

dorssel, 2023-11-07:
See https://github.com/dorssel/usbipd-win/issues/558#issuecomment-1457342674

Tapir28, 2023-11-08:
[log.zip](https://github.com/dorssel/usbipd-win/files/13295335/log.zip)

Thank you, that was the issue, please find the capture and powershell output attached.

dorssel, 2023-11-08:
The problem does not appear to be USB, but rather the network communication between WSL and the host. Currently *something* is killing the network connection; could be either WSL/Linux or Windows side, maybe a 3rd party firewall? You could try without the experimental mirror mode...

thijsputman, 2023-11-08:
Chances are high it works fine without the ([not experimental any more 😉](https://github.com/microsoft/WSL/releases/tag/2.0.5)) `mirrored`-mode... 😇

Upgraded WSL2 to `2.0.7` today (from `1.3.x`) and enabled (amongst other things) the new `mirrored` networking-mode.

Prior to the upgrade, connecting my Yubikey into WSL2 worked fine. After the upgrade (and enabling mirrored networking) it still works fine, but "reliably" disconnects after 20-seconds:

```log
[ 1685.220228] input: Yubico YubiKey OTP+FIDO+CCID as /devices/platform/vhci_hcd.0/usb1/1-1/1-1:1.0/0003:1050:0407.0009/input/input4
[ 1685.290973] hid-generic 0003:1050:0407.0009: input: USB HID v1.10 Keyboard [Yubico YubiKey OTP+FIDO+CCID] on usb-vhci_hcd.0-1/input0
...
[ 1705.858034] vhci_hcd: connection reset by peer
[ 1705.858322] vhci_hcd: stop threads
[ 1705.858335] vhci_hcd: release socket
[ 1705.858341] vhci_hcd: disconnect device
```

Tried it many times and the time frame in the `dmesg` output is always 20-seconds – guess that's not a coincidence but a 20-second timeout somewhere?

Have some other obligations right now, but I'll try to confirm tonight/tomorrow that the issue is resolved by disabling the mirrored networking-mode...

dorssel, 2023-11-08:
@thijsputman 
OK, with mirrored mode not being experimental anymore, `usbipd-win` should add support quickly. Will do...

The 20 seconds is a real smell: `usbipd-win` uses a TCP keep-alive timeout of 10 seconds, with retries every 0.5 seconds. Total timeout expected is 15 seconds (docs state 10 retries), but it could very well be 20 seconds (given that the number of retries may be OS dependent; docs are not quite clear).

It could be that WSL mirrored mode does not honor keepalives (which would be a bug of WSL), or maybe that loopback connections never support it (I have to look that up). With mirrored mode, is it correct that you use 127.0.0.1 as remote? Could you try with the local IP address (not loopback...)?

thijsputman, 2023-11-08:
Correct, the remote is configured as `127.0.0.1`.

With mirrored networking-mode enabled, there doesn't appear to be an (obvious) way to connect to the remote _without_ using a loopback connection (that is, if I'm properly understanding what's happening in the background)...

Using the remote's regular local IP address results in a "System Error" when trying to connect.

Adding `experimental.hostAddressLoopback=true` to the WSL configuration ([see documentation](https://learn.microsoft.com/en-us/windows/wsl/wsl-config#experimental-configuration-settings)) makes the local IP address work, but that then also disconnects after 20-seconds. That and the way I read the documentation seems to hint that it's a loopback connection also somehow?

The mirrored networking hoists all active (ie, connected) network adapters from Windows into WSL2. They're fully "shared" on both sides – if something on the Windows-side is listening on a port, that port cannot be used in WSL2.

dorssel, 2023-11-08:
Thanks for the info. Yes, `experimental.hostAddressLoopback=true` indeed makes all connections to adapter IP addresses of the host effectively using the loopback interface instead. So it makes sense you see the same behavior then.

I found this: https://stackoverflow.com/questions/30943686/is-any-kind-of-keepalive-necessary-on-localhost-socket
It discusses an obscure scenario where even on loopback, keep-alives make sense. And I did not find anywhere that keep-alives on loopback behave differently, or are not supported. I will test the keep-alive behavior of WSL mirrored mode, just to confirm or rule it out....

dorssel, 2023-11-12:
Reported: https://github.com/microsoft/WSL/issues/10741

closed 2023-12-06 by dorssel