dorssel/usbipd-win #752: can't attach device from WSL

ZZH-Finzlize, 2023-11-07:
When I use command `usbip list -r 192.168.1.3`, I found it can list the devices that bind with usbipd-win, but I can't use `usbip attach -r 192.168.1.3 -b 4-2` to attach this device to WSL. Why is that? I think the connection should be able to build from both side (from Windows or from WSL)

[image]

[image]

dhcstruggle, 2023-11-07:
Same issue here.

ZZH-Finzlize, 2023-11-07:
> I found solution here and it works for me. just FYI: [#663 (comment)](https://github.com/dorssel/usbipd-win/issues/663#issuecomment-1683761470)

It is different than my issue, because I can build the connection from Windows use `usbipd wsl attach` command, this command works well, but what I want is build this connection from WSL by `usbip attach` command, and this command will report "Device busy". 

Still thx for that.

dorssel, 2023-11-07:
"Device busy" means that the host is currently actively using the device. Either stop the software/service on Windows that is using the device, or use `usbipd bind --force`.

ZZH-Finzlize, 2023-11-08:
> "Device busy" means that the host is currently actively using the device. Either stop the software/service on Windows that is using the device, or use `usbipd bind --force`.

@dorssel Actually, That device is already bind to usbipd with the --force flag, but it just can't attach from WSL (from Windows is OK), and I find if I change `networkingMode` to `mirrored`, then I can attach the same device from WSL, meanwhile, from Windows to attach the device will fail.

[image]

.wslconfig
[image]

ZZH-Finzlize, 2023-11-08:
> "Device busy" means that the host is currently actively using the device. Either stop the software/service on Windows that is using the device, or use `usbipd bind --force`.

I guess I find the reason, attach a device from WSL has to be under the root privilege, if execute the `usbip` command without `sudo`, even chang the .wslconfig is useless.

dorssel, 2023-11-08:
> attach a device from WSL has to be under the root privilege

That is correct, and that is what `usbipd wsl attach` uses underwater. mirror mode is not yet supported, see https://github.com/dorssel/usbipd-win/issues/714. Is your issue resolved now?

ZZH-Finzlize, 2023-11-08:
> > attach a device from WSL has to be under the root privilege
> 
> That is correct, and that is what `usbipd wsl attach` uses underwater. mirror mode is not yet supported, see #714. Is your issue resolved now?

Yes, Thanks for the help.

closed 2023-11-08 by dorssel