dorssel/usbipd-win #755: Unable to attach nRF52840 MDK to WSL2

dekimsey, 2023-11-11:
I was trying to connect my [nRF52840 MDK](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/) (with the nrfConnect [firmware](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/main/firmware/ble_connectivity) loaded) in my WSL2 VM but I'm seeing the device connect briefly and then promptly drop within 2-3 seconds.

I've attached the various logs I can find following the troubleshooting doc.

Version (`usbipd --version`):
```
3.2.0+38.Branch.master.Sha.7ad7b9904ac0db8d759e3a53a22b1126da01b2be
```

WSL (`uname -a`):
```
Linux DESKTOP-LF0L3DD 5.15.90.1-microsoft-standard-WSL2 #1 SMP Fri Jan 27 02:56:13 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
```

Devices (`usbipd wsl list`)
```
BUSID  VID:PID    DEVICE                                                        STATE
1-7    1915:c00a  nRF Connect USB CDC ACM (COM4)                                Not attached
...
```

### Logs

<details><summary>dmesg</summary>
<p>

```
[  730.001504] usb 1-1: new full-speed USB device number 2 using vhci_hcd
[  730.091506] vhci_hcd: vhci_device speed not set
[  730.161493] usb 1-1: SetAddress Request (2) to port 0
[  730.287618] usb 1-1: config 1 has an invalid interface number: 2 but max is 1
[  730.288313] usb 1-1: config 1 has no interface number 0
[  730.300718] usb 1-1: New USB device found, idVendor=1915, idProduct=c00a, bcdDevice= 1.00
[  730.301550] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  730.302288] usb 1-1: Product: nRF52 Connectivity
[  730.302611] usb 1-1: Manufacturer: Nordic Semiconductor
[  730.302964] usb 1-1: SerialNumber: D5D921902F15
[  730.655434] vhci_hcd: connection closed
[  730.655583] vhci_hcd: stop threads
[  730.656319] vhci_hcd: release socket
[  730.656607] vhci_hcd: disconnect device
[  735.361502] usb 1-1: can't set config #1, error -110
[  735.362013] usb 1-1: USB disconnect, device number 2
```

</p></details>

<details>
<summary>usbipd server logs</summary>
<p>

```
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = 'C:\serial.pcapng'
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:SnapLength = unlimited
dbug: Usbipd.Server[1000]
      usbipd:Port = 3240
dbug: Usbipd.Server[1000]
      3.2.0+38.Branch.master.Sha.7ad7b9904ac0db8d759e3a53a22b1126da01b2be
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveInterval = 500 ms
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveTime = 10000 ms
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Program Files\usbipd-win\
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.PcapNg[1000]
      Flushing
dbug: Usbipd.Server[1000]
      new connection from 172.29.242.27
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.29.242.27
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 416 ms
info: Usbipd.ConnectedClient[1]
      Client 172.29.242.27 claimed device at 1-7 (USB\VID_1915&PID_C00A\D5D921902F15).
dbug: 1-7.0[1000]
      Endpoint created
dbug: 1-7.0[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: 1-7.0[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: 1-7.0[1000]
      USBSUP_XFER_STALL -> EPIPE -> -32
dbug: 1-7.0[1000]
      Endpoint created
dbug: 1-7.0[1000]
      Trapped SET_CONFIGURATION: 1
info: Usbipd.ConnectedClient[2]
      Client 172.29.242.27 released device at 1-7 (USB\VID_1915&PID_C00A\D5D921902F15).
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
fail: Usbipd.ConnectedClient[3]
      An exception occurred while communicating with the client:
      System.ComponentModel.Win32Exception (87): DeviceIoControl returned error ERROR_INVALID_PARAMETER
         at Usbipd.AttachedEndpoint.HandleSubmitAsync(UsbIpHeaderBasic, UsbIpHeaderCmdSubmit, CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\AttachedEndpoint.cs:line 288
         at Usbipd.AttachedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\AttachedClient.cs:line 135
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 229
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 252
         at Usbipd.ConnectedClient.HandleRequestImportAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 271
         at Usbipd.ConnectedClient.RunAsync(CancellationToken) in D:\a\usbipd-win\usbipd-win\Usbipd\ConnectedClient.cs:line 58
dbug: Usbipd.Server[1000]
      connection close: DeviceIoControl returned error ERROR_INVALID_PARAMETER
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.PcapNg[1000]
      Flushing
info: Microsoft.Hosting.Lifetime[0]
      Application is shutting down...
dbug: Microsoft.Extensions.Hosting.Internal.Host[3]
      Hosting stopping
dbug: Microsoft.Extensions.Hosting.Internal.Host[4]
      Hosting stopped
```

</p>
</details>

And a pcap as well, [serial.zip](https://github.com/dorssel/usbipd-win/files/13324316/serial.zip)

Thank you!

dorssel, 2023-11-11:
The device "unplugs" right after setting the configuration. This usually means that the device resets, which to Windows appears as an unplug/replug event. Maybe the device even comes back with a different VID:PID. Like it is first in a "boot mode" and then after configuring it becomes a "different" device. Can you do a `usbipd list` right before attach, and right after? To see if the VID:PID actually changes?

dorssel, 2023-12-06:
stale

closed 2023-12-06 by dorssel