dorssel/usbipd-win #758: usbip no longer working

Bestagon-42, 2023-11-13:
Hello,
I am attempting to pass in a game controller into WSL2 Ubuntu-22.04. I installed usbipd yesterday and was able to pass in a controller successfully, however due to an unrelated issue that controller didn't work for what I was attempting.

Today I went to try again with a generic XBox controller plugged into my PC. and received the following error:

Command in Windows Powershell:
`usbipd wsl attach --busid 1-8`

Result:
`usbipd: error: WSL 'usbip' client not correctly installed. See https://github.com/dorssel/usbipd-win/wiki/WSL-support for the latest instructions.`

did some poking around but I am not sure what the problem is, running `usbip` within WSL2 returns the following:

`WARNING: usbip not found for kernel 5.15.90.1-microsoft

  You may need to install the following packages for this specific kernel:
    linux-tools-5.15.90.1-microsoft-standard-WSL2
    linux-cloud-tools-5.15.90.1-microsoft-standard-WSL2

  You may also want to install one of the following packages to keep up to date:
    linux-tools-standard-WSL2
    linux-cloud-tools-standard-WSL2`

Running the [install commands](https://github.com/dorssel/usbipd-win/wiki/WSL-support#usbip-client-tools) returns the following:

Command:
`sudo apt install linux-tools-virtual hwdata`

Result:
`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
hwdata is already the newest version (0.357-1).
linux-tools-virtual is already the newest version (5.15.0.88.85).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.`

Command:
`sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20`

Result:
`update-alternatives: error: alternative path /usr/lib/linux-tools/5.15.0-88-generic/usbip doesn't exist`

Any help would be greatly appreciated!

Version information:

WSL2 Kernel: 5.15.90.1-microsoft-standard-WSL2
Ubuntu: Ubuntu 22.04.3 LTS
Windows: Windows 10 Home 22H2

dorssel, 2023-11-14:
What is the output of `ls -l /usr/lib/linux-tools/*/usbip` ?

Bestagon-42, 2023-11-14:
Output:
```
lrwxrwxrwx 1 root root 33 Nov 13 12:34 /usr/lib/linux-tools/5.15.0-88-generic/usbip -> ../../linux-tools-5.15.0-88/usbip
```

dorssel, 2023-11-14:
And `ls -l /usr/lib/linux-tools-5.15.0-88/usbip` ?

Bestagon-42, 2023-11-14:
Output:
```
ls: cannot access '/usr/lib/linux-tools-5.15.0-88/usbip': No such file or directory
```

dorssel, 2023-11-14:
For some reason, you're `usbip` isn't installed anymore. Try reinstalling with:

```
sudo apt remove linux-tools-virtual --autoremove
sudo apt install linux-tools-virtual
```

and then do the `update-alternatives` again.

Bestagon-42, 2023-11-14:
Still getting an alternative path doesn't exist error from the 'update-alternatives' command

```
brad@BL-Home-System:~$ sudo apt remove linux-tools-virtual --autoremove
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages will be REMOVED:
  linux-tools-virtual
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 20.5 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 147102 files and directories currently installed.)
Removing linux-tools-virtual (5.15.0.88.85) ...
brad@BL-Home-System:~$ sudo apt install linux-tools-virtual
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  linux-tools-virtual
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 2360 B of archives.
After this operation, 20.5 kB of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu jammy-updates/main amd64 linux-tools-virtual amd64 5.15.0.88.85 [2360 B]
Fetched 2360 B in 0s (7094 B/s)
Selecting previously unselected package linux-tools-virtual.
(Reading database ... 147099 files and directories currently installed.)
Preparing to unpack .../linux-tools-virtual_5.15.0.88.85_amd64.deb ...
Unpacking linux-tools-virtual (5.15.0.88.85) ...
Setting up linux-tools-virtual (5.15.0.88.85) ...
brad@BL-Home-System:~$ sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20
update-alternatives: error: alternative path /usr/lib/linux-tools/5.15.0-88-generic/usbip doesn't exist
```

dorssel, 2023-11-14:
`linux-tools-virtual` is supposed to install much more : `libnuma1 libpci3 linux-tools-5.15.0-88 linux-tools-5.15.0-88-generic linux-tools-common` as well...

Something is broken in your package system. For now, just try to install the dependency directly:

```
sudo apt install linux-tools-5.15.0-88 linux-tools-5.15.0-88-generic
```

(these should have been installed automagically by `linux-tools-virtual`). And then the `update-alternatives` again. Don't know what is wrong, but this should get it working again.
```

Bestagon-42, 2023-11-14:
I really appreciate the help! had to run `sudo apt remove linux-tools-5.15.0-88 linux-tools-5.15.0-88-generic` and then `sudo apt install linux-tools-5.15.0-88 linux-tools-5.15.0-88-generic` and that fixed it

closed 2023-11-14 by Bestagon-42