dorssel/usbipd-win #764: USB Ethernet adapters

taufderl, 2023-11-14:
I'd love to get usbipd working with USB Ethernet adapters. I have several lying around and tested them all with a negative result. I have documented the errors below for two of my adapters:

1. Linksys USB3GIG V1 (USB A)
2. Dell USB C Ethernet Adapter (no exact model label)

I compiled my own WSL kernel (based on [github.com/microsoft/WSL2-Linux-Kernel](https://github.com/microsoft/WSL2-Linux-Kernel) to include the necessary drivers. I may test this with an HyperV VM in the next days to ensure that the kernel is not the issue here. Considering the devices appear correctly in WSL, I don't think it is.
#### usbipd, WSL and Kernel versions:

```
PS C:\> usbipd.exe --version
3.2.0+38.Branch.master.Sha.7ad7b9904ac0db8d759e3a53a22b1126da01b2be

PS C:\> wsl --version
WSL version: 1.2.5.0
Kernel version: 5.15.90.1
WSLg version: 1.0.51
MSRDC version: 1.2.3770
Direct3D version: 1.608.2-61064218
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.22631.2428

PS C:\> wsl -l -v
  NAME                   STATE           VERSION
  Development            Running         2
  <...>
```

```
user@development:~$ uname -a
Linux development 5.15.133.1-microsoft-standard-WSL2+ #2 SMP Thu Nov 2 11:59:12 CET 2023 x86_64 x86_64 x86_64 GNU/Linux
```

## Linksys USB3GIG V1

Attach the device

```
PS C:\> usbipd.exe wsl list
7-13   13b1:0041  Linksys USB3GIGV1                                             Not attached
PS C:\> usbipd.exe wsl attach -b 7-13 -d Development
```

Confirm it is being recognized:

```
user@development:~$ dmesg
<...>
[583232.079012] vhci_hcd vhci_hcd.0: pdev(0) rhport(1) sockfd(3)
[583232.079018] vhci_hcd vhci_hcd.0: devid(458765) speed(5) speed_str(super-speed)
[583232.079060] vhci_hcd vhci_hcd.0: Device attached
[583232.431558] usb 2-2: SetAddress Request (3) to port 1
[583232.431565] usb 2-2: new SuperSpeed USB device number 3 using vhci_hcd
[583232.473212] usb 2-2: New USB device found, idVendor=13b1, idProduct=0041, bcdDevice=30.00
[583232.473215] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[583232.473217] usb 2-2: Product: Linksys USB3GIGV1
[583232.473218] usb 2-2: Manufacturer: Linksys
[583232.473218] usb 2-2: SerialNumber: 000001000000
[583232.479250] cdc_ether 2-2:2.0 eth1: register 'cdc_ether' at usb-vhci_hcd.0-2, CDC Ethernet Device, 60:38:e0:e3:84:6c
[583232.487114] cdc_ether 2-2:2.0 enx6038e0e3846c: renamed from eth1
```

```
user@development:~$ lsusb
Bus 002 Device 003: ID 13b1:0041 Linksys Gigabit Ethernet Adapter
```

```
user@development:~$ ip a
<...>
4: enx6038e0e3846c: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 60:38:e0:e3:84:6c brd ff:ff:ff:ff:ff:ff
```

Let's try to bring the device up and monitor traffic:

```
user@development:~$ sudo ip link set enx6038e0e3846c up
user@development:~$ sudo tcpdump -n -i enx6038e0e3846c
```

In another terminal, request an IP address:

```
user@development:~$ sudo dhclient enx6038e0e3846c
```

This results in the requests being shown in tcpdump, however nothing comes back in (the cable is connected to a router):

```
user@development:~$ sudo tcpdump -n -i enx6038e0e3846c
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on enx6038e0e3846c, link-type EN10MB (Ethernet), snapshot length 262144 bytes
22:54:20.989074 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 60:38:e0:e3:84:6c, length 300
22:54:23.675546 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 60:38:e0:e3:84:6c, length 300
22:54:29.255936 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 60:38:e0:e3:84:6c, length 300
22:54:40.659316 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 60:38:e0:e3:84:6c, length 300
22:54:54.466115 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 60:38:e0:e3:84:6c, length 300
22:55:03.896568 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 60:38:e0:e3:84:6c, length 300
```

It takes some time, but after a minute or so I see lots of errors in `dmesg` like this:

```
[584020.874905] vhci_hcd: unlink->seqnum 33806
[584020.874909] vhci_hcd: urb->status -104
[584020.875172] vhci_hcd: unlink->seqnum 33807
[584020.875176] vhci_hcd: urb->status -104
[584020.875301] vhci_hcd: unlink->seqnum 33808
[584020.875304] vhci_hcd: urb->status -104
<...>
[584020.886442] vhci_hcd: unlink->seqnum 34103
[584020.886442] vhci_hcd: urb->status -104
[584020.886443] vhci_hcd: unlink->seqnum 34104
[584020.886444] vhci_hcd: urb->status -104
[584020.886450] vhci_hcd: unlink->seqnum 34105
[584020.886450] vhci_hcd: urb->status -104
[584269.821211] usb 2-2: seqnum max
```

I repeated this and also tested without running `tcpdump` just to be sure that it is not the `promiscuous mode` breaking things. Always the same result.
## Dell USB C Ethernet Adapter

Attach the device

```
PS C:\> usbipd.exe wsl list
BUSID  VID:PID    DEVICE
1-2    0bda:8153  Realtek USB GbE Family Controller #2                          Not attached
PS C:\> usbipd.exe wsl attach -b 1-2 -d Development
```

Confirm it is being recognized:

```
user@development:~$ dmesg
<...>
[583206.137590] vhci_hcd vhci_hcd.0: pdev(0) rhport(0) sockfd(3)
[583206.137689] vhci_hcd vhci_hcd.0: devid(65538) speed(5) speed_str(super-speed)
[583206.137839] vhci_hcd vhci_hcd.0: Device attached
[583206.491553] usb 2-1: SetAddress Request (2) to port 0
[583206.491623] usb 2-1: new SuperSpeed USB device number 2 using vhci_hcd
[583206.557452] usb 2-1: New USB device found, idVendor=0bda, idProduct=8153, bcdDevice=30.00
[583206.557460] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=6
[583206.557463] usb 2-1: Product: USB 10/100/1000 LAN
[583206.557528] usb 2-1: Manufacturer: Realtek
[583206.557529] usb 2-1: SerialNumber: 000001000000
[583206.565763] r8153_ecm 2-1:2.0 eth1: register 'r8153_ecm' at usb-vhci_hcd.0-1, RTL8153 ECM Device, 9c:eb:e8:74:c8:d1
[583206.582070] r8153_ecm 2-1:2.0 enx9cebe874c8d1: renamed from eth1
```

```
user@development:~$ lsusb
Bus 002 Device 002: ID 0bda:8153 Realtek Semiconductor Corp. RTL8153 Gigabit Ethernet Adapter
```

```
user@development:~$ ip a
<...>
3: enx9cebe874c8d1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 9c:eb:e8:74:c8:d1 brd ff:ff:ff:ff:ff:ff
```

Bring the device up and monitor traffic:

```
user@development:~$ sudo ip link set enx9cebe874c8d1 up
user@development:~$ sudo tcpdump -n -i enx9cebe874c8d1
```

After some time without incoming traffic and watching the DHCP requests:

```
user@development:~$ sudo tcpdump -n -i enx9cebe874c8d1
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on enx9cebe874c8d1, link-type EN10MB (Ethernet), snapshot length 262144 bytes
23:05:01.217948 IP6 fe80::9eeb:e8ff:fe74:c8d1 > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
23:05:01.317726 IP6 fe80::9eeb:e8ff:fe74:c8d1 > ff02::16: HBH ICMP6, multicast listener report v2, 1 group record(s), length 28
23:05:10.848603 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 9c:eb:e8:74:c8:d1, length 300
23:05:13.475915 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 9c:eb:e8:74:c8:d1, length 300
23:05:15.797707 IP6 fe80::9eeb:e8ff:fe74:c8d1 > ff02::2: ICMP6, router solicitation, length 16
23:05:17.557711 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 9c:eb:e8:74:c8:d1, length 300
23:05:22.889465 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 9c:eb:e8:74:c8:d1, length 300
23:05:31.321460 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 9c:eb:e8:74:c8:d1, length 300
23:05:46.940620 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 9c:eb:e8:74:c8:d1, length 300
```

the same errors in `dmesg` appear:

```
[584732.410324] vhci_hcd: unlink->seqnum 113370
[584732.410373] vhci_hcd: urb->status -104
[584732.410409] vhci_hcd: unlink->seqnum 113371
[584732.410436] vhci_hcd: urb->status -104
[584732.410441] vhci_hcd: unlink->seqnum 113372
[584732.410470] vhci_hcd: urb->status -104
<...>
[584732.422432] vhci_hcd: unlink->seqnum 113667
[584732.422434] vhci_hcd: urb->status -104
[584732.422530] vhci_hcd: unlink->seqnum 113668
[584732.422533] vhci_hcd: urb->status -104
[584732.422546] vhci_hcd: unlink->seqnum 113669
[584732.422547] vhci_hcd: urb->status -104
```

## Conclusion

All my Ethernet adapters currently do not work with `usbipd` and throw the documented errors. I am happy to debug more, provide logs and more device details if necessary.

dorssel, 2023-11-15:
Hi Tim!

Can you follow the guide in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting and get some console logging and USB captures? Other people have made Bluetooth adapters work, even with networking, so I see no reason why we wouldn't get this to work.

taufderl, 2023-11-16:
Hi Frans!

thanks a lot! Here are the log and pcap file for the Linksys device:

[Linksys_13b1-0041.log](https://github.com/dorssel/usbipd-win/files/13377569/Linksys_13b1-0041.log)
[Linksys_13b1-0041.pcapng.zip](https://github.com/dorssel/usbipd-win/files/13377579/Linksys_13b1-0041.pcapng.zip)