dorssel/usbipd-win #770: usbip: error: import device

elcapitantommy, 2023-11-18:
Hi all

I've used usbipd to connect  to WSL using command prompt for many months but it's suddenly stopped working and gives a "usbip: error: import device" error.

`C:\Windows\System32>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-5    27c6:639c  Goodix MOC Fingerprint                                        Not shared
2-6    0c45:6739  Integrated Webcam                                             Not shared
2-10   8087:0033  Intel(R) Wireless Bluetooth(R)                                Not shared
4-1    1a86:7523  USB-SERIAL CH340 (COM12)                                      Shared
5-1    1a86:7523  USB-SERIAL CH340 (COM13)                                      Shared
6-2    2109:0103  USB 2.0 BILLBOARD                                             Not shared
6-3    1a86:7523  USB-SERIAL CH340 (COM10)                                      Shared

Persisted:
GUID                                  DEVICE
af985450-94f5-4145-bd97-2831e6c2058d  USB-SERIAL CH340 (COM9)

C:\Windows\System32>usbipd wsl attach --busid 5-1
usbip: error: import device
usbipd: error: Failed to attach device with busid '5-1'.`

I'm on version 3.2.0 of usbipd-win and the latest version of WSL2.

Through trial and error (I don't know what I'm doing) I've got it to work doing the following, just in case anyone else is having this problem and is struggling to get things working:

1.` ipconfig` in command prompt to get the IP address used by WSL (Ethernet adapter vEthernet (WSL))
2. in WSL `usbip list -r <IP address>`
3. in WSL `sudo usbip attach -r <IP> -b X-X` (replacing the Xs with the busid)

Everything appears to now be working fine, I'm using three 3D printers and upon connecting the third the first two disconnected and I had to detach them in WSL, unplug and then reconnect the printers, and the attach again in WSL - things now seem stable at the moment.

I'm sorry if this is the wrong place to raise this but I couldn't find an explanation/cause/resolution online and I don't know if I'm going to have stability issues using this new process.

If anyone can shed any light on this I'm curious as to what is causing this error.

If you would like more info please let me know.

Thanks

Tom

dorssel, 2023-11-18:
Try running with full logging, as in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

elcapitantommy, 2023-11-18:
> Try running with full logging, as in https://github.com/dorssel/usbipd-win/wiki/Troubleshooting.

Thanks for getting back to me, this is what I got when trying to attach a device:

`C:\Windows\System32>usbipd server Logging:LogLevel:Default=Trace
dbug: Microsoft.Extensions.Hosting.Internal.Host[1]
      Hosting starting
dbug: Usbipd.PcapNg[1000]
      usbipd:PcapNg:Path = ''
dbug: Usbipd.Server[1000]
      usbipd:Port = 3240
dbug: Usbipd.Server[1000]
      3.2.0+38.Branch.master.Sha.7ad7b9904ac0db8d759e3a53a22b1126da01b2be
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveInterval = 500 ms
dbug: Usbipd.Server[1000]
      usbipd:TcpKeepAliveTime = 10000 ms
info: Microsoft.Hosting.Lifetime[0]
      Application started. Press Ctrl+C to shut down.
info: Microsoft.Hosting.Lifetime[0]
      Hosting environment: Production
info: Microsoft.Hosting.Lifetime[0]
      Content root path: C:\Program Files\usbipd-win\
dbug: Microsoft.Extensions.Hosting.Internal.Host[2]
      Hosting started
dbug: Usbipd.Server[1000]
      new connection from 172.22.122.235
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed
dbug: Usbipd.Server[1000]
      new connection from 172.22.122.235
dbug: Usbipd.ConnectedClient[1000]
      Received opcode: OP_REQ_IMPORT
dbug: Usbipd.ConnectedClient[1000]
      Claiming took 224 ms
info: Usbipd.ConnectedClient[1]
      Client 172.22.122.235 claimed device at 3-1 (USB\VID_1A86&PID_7523\6&13145193&0&1).
info: Usbipd.ConnectedClient[2]
      Client 172.22.122.235 released device at 3-1 (USB\VID_1A86&PID_7523\6&13145193&0&1).
dbug: Usbipd.ConnectedClient[1000]
      Unbind or unplug while attached
dbug: Usbipd.Server[1000]
      connection close: Unable to read beyond the end of the stream.
dbug: Usbipd.Server[1000]
      connection closed`

dorssel, 2023-11-18:
> Unbind or unplug while attached

That is literally what is happening. Either the device resets (which is seen by Windows as an unplug/replug), or Linux hangs up. Probably the latter. what are the messages in syslog/dmesg ?

elcapitantommy, 2023-11-18:
So nothing is appearing in dmesg when I try to attach a device but my syslog looks like this:

Nov 18 22:05:09 TDixon systemd[1]: Starting User Manager for UID 1000...
Nov 18 22:05:09 TDixon systemd[17642]: Queued start job for default target Main User Target.
Nov 18 22:05:09 TDixon systemd[17642]: Created slice User Application Slice.
Nov 18 22:05:09 TDixon systemd[17642]: Reached target Paths.
Nov 18 22:05:09 TDixon systemd[17642]: Reached target Timers.
Nov 18 22:05:09 TDixon systemd[17642]: Listening on GnuPG network certificate management daemon.
Nov 18 22:05:09 TDixon systemd[17642]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Nov 18 22:05:09 TDixon systemd[17642]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Nov 18 22:05:09 TDixon systemd[17642]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Nov 18 22:05:09 TDixon systemd[17642]: Listening on GnuPG cryptographic agent and passphrase cache.
Nov 18 22:05:09 TDixon systemd[17642]: Listening on debconf communication socket.
Nov 18 22:05:09 TDixon systemd[17642]: Listening on REST API socket for snapd user session agent.
Nov 18 22:05:09 TDixon systemd[17642]: Reached target Sockets.
Nov 18 22:05:09 TDixon systemd[17642]: Reached target Basic System.
Nov 18 22:05:09 TDixon systemd[17642]: Reached target Main User Target.
Nov 18 22:05:09 TDixon systemd[17642]: Startup finished in 27ms.
Nov 18 22:05:09 TDixon systemd[1]: Started User Manager for UID 1000.
Nov 18 22:05:09 TDixon systemd[1]: Started Session c104 of User tdixon.
Nov 18 22:05:09 TDixon systemd[1]: session-c104.scope: Deactivated successfully.
Nov 18 22:05:09 TDixon systemd[1]: Started Session c105 of User tdixon.
Nov 18 22:05:09 TDixon systemd[1]: session-c105.scope: Deactivated successfully.
Nov 18 22:05:09 TDixon systemd[1]: Started Session c106 of User tdixon.
Nov 18 22:05:09 TDixon systemd[1]: session-c106.scope: Deactivated successfully.
Nov 18 22:05:10 TDixon systemd[1]: Started Session c107 of User tdixon.
Nov 18 22:05:10 TDixon systemd[1]: session-c107.scope: Deactivated successfully.
Nov 18 22:05:10 TDixon systemd[1]: Started Session c108 of User tdixon.
Nov 18 22:05:10 TDixon systemd[1]: session-c108.scope: Deactivated successfully.
Nov 18 22:05:13 TDixon python[219]: [websockets.py:on_close()] - Websocket Closed: ID: 140056316261232 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.33
Nov 18 22:05:13 TDixon python[219]: [websockets.py:on_close()] - Websocket Closed: ID: 140056316261040 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.31
Nov 18 22:05:13 TDixon python[219]: [websockets.py:on_close()] - Websocket Closed: ID: 140056316067792 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.30
Nov 18 22:05:13 TDixon python[218]: [websockets.py:on_close()] - Websocket Closed: ID: 140291329935680 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.30
Nov 18 22:05:13 TDixon python[218]: [websockets.py:on_close()] - Websocket Closed: ID: 140291340152704 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.33
Nov 18 22:05:13 TDixon python[218]: [websockets.py:on_close()] - Websocket Closed: ID: 140291329936160 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.32
Nov 18 22:05:13 TDixon python[220]: [websockets.py:on_close()] - Websocket Closed: ID: 139823385833200 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.33
Nov 18 22:05:13 TDixon python[220]: [websockets.py:on_close()] - Websocket Closed: ID: 139823385884704 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.30
Nov 18 22:05:13 TDixon python[220]: [websockets.py:on_close()] - Websocket Closed: ID: 139823385651200 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 5.32
Nov 18 22:05:13 TDixon python[220]: [websockets.py:on_close()] - Websocket Closed: ID: 139823385675904 Close Code: 1001, Close Reason: None, Pong Time Elapsed: 8.00
Nov 18 22:05:20 TDixon systemd[1]: Stopping User Manager for UID 1000...
Nov 18 22:05:20 TDixon systemd[17642]: Stopped target Main User Target.
Nov 18 22:05:20 TDixon systemd[17642]: Stopped target Basic System.
Nov 18 22:05:20 TDixon systemd[17642]: Stopped target Paths.
Nov 18 22:05:20 TDixon systemd[17642]: Stopped target Sockets.
Nov 18 22:05:20 TDixon systemd[17642]: Stopped target Timers.
Nov 18 22:05:20 TDixon systemd[17642]: Closed GnuPG network certificate management daemon.
Nov 18 22:05:20 TDixon systemd[17642]: Closed GnuPG cryptographic agent and passphrase cache (access for web browsers).
Nov 18 22:05:20 TDixon systemd[17642]: Closed GnuPG cryptographic agent and passphrase cache (restricted).
Nov 18 22:05:20 TDixon systemd[17642]: Closed GnuPG cryptographic agent (ssh-agent emulation).
Nov 18 22:05:20 TDixon systemd[17642]: Closed GnuPG cryptographic agent and passphrase cache.
Nov 18 22:05:20 TDixon systemd[17642]: Closed debconf communication socket.
Nov 18 22:05:20 TDixon systemd[17642]: Closed REST API socket for snapd user session agent.
Nov 18 22:05:20 TDixon systemd[17642]: Removed slice User Application Slice.
Nov 18 22:05:20 TDixon systemd[17642]: Reached target Shutdown.
Nov 18 22:05:20 TDixon systemd[17642]: Finished Exit the Session.
Nov 18 22:05:20 TDixon systemd[17642]: Reached target Exit the Session.
Nov 18 22:05:20 TDixon systemd[1]: user@1000.service: Deactivated successfully.
Nov 18 22:05:20 TDixon systemd[1]: Stopped User Manager for UID 1000.
Nov 18 22:05:20 TDixon systemd[1]: Stopping User Runtime Directory /run/user/1000...
Nov 18 22:05:20 TDixon systemd[1]: run-user-1000.mount: Deactivated successfully.
Nov 18 22:05:20 TDixon systemd[1]: user-runtime-dir@1000.service: Deactivated successfully.
Nov 18 22:05:20 TDixon systemd[1]: Stopped User Runtime Directory /run/user/1000.
Nov 18 22:05:20 TDixon systemd[1]: Removed slice User Slice of UID 1000.

dorssel, 2023-11-18:
There is nothing in that syslog about USB, yet:

> Client 172.22.122.235 claimed device at 3-1 (USB\VID_1A86&PID_7523\6&13145193&0&1).

This indicates that `usbip` has run at the Linux side. Are you using the right distribution to see the logging? What is the output of `wsl --list --verbose`?

elcapitantommy, 2023-11-18:
C:\Windows\System32>wsl --list --verbose
  NAME      STATE           VERSION
* Ubuntu    Running         2

dorssel, 2023-11-29:
Are you running Windows 11 or 10? WSL has changed quite a bit, especially for Windows 11. Soon, version 4.0.0 will be released that should solve these problems (see https://github.com/dorssel/usbipd-win/discussions/774). Until then, manually run `usbip` from WSL as a workaround.

dorssel, 2023-11-29:
WSL is changing all the time, more so for Windows 11 than for Windows 10. It *could* explain why things were working before, and now they are not.

dorssel, 2023-12-06:
4.0.0 has been release, which improves WSL support. Does that resolve your issue?

elcapitantommy, 2023-12-06:
No it hasn't, still getting the WSL usbip: error: import device message

dorssel, 2023-12-11:
We really need to get the logging from Linux. Maybe try `dmesg -w` instead of syslog while attaching.

Version 4.0.0 will display which distro and IP address is being used for attaching. Does that match what you use to manually attach?

elcapitantommy, 2023-12-11:
So my CMD output looks like this:

```
C:\Windows\System32>usbipd attach --wsl --busid 2-1
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.22.112.1 to reach the host.
WSL usbip: error: import device
usbipd: error: Failed to attach device with busid '2-1'
```

And my dmesg -w output is:
```
[   72.036204] hv_pci 6d836b00-d4a8-4950-968b-51ea909d7298: PCI VMBus probing: Using version 0x10004
[   72.063457] 9pnet_virtio: no channels available for device drvfsa
[   72.063958] hv_pci 6d836b00-d4a8-4950-968b-51ea909d7298: PCI host bridge to bus d4a8:00
[   72.064082] pci_bus d4a8:00: root bus resource [mem 0xbffe10000-0xbffe12fff window]
[   72.064083] pci_bus d4a8:00: No busn resource found for root bus, will use [bus 00-ff]
[   72.064134] WSL (1) WARNING: mount: waiting for virtio device drvfsa
[   72.064557] pci d4a8:00:00.0: [1af4:1049] type 00 class 0x010000
[   72.065022] pci d4a8:00:00.0: reg 0x10: [mem 0xbffe10000-0xbffe10fff 64bit]
[   72.065341] pci d4a8:00:00.0: reg 0x18: [mem 0xbffe11000-0xbffe11fff 64bit]
[   72.065633] pci d4a8:00:00.0: reg 0x20: [mem 0xbffe12000-0xbffe12fff 64bit]
[   72.067843] pci_bus d4a8:00: busn_res: [bus 00-ff] end is updated to 00
[   72.067849] pci d4a8:00:00.0: BAR 0: assigned [mem 0xbffe10000-0xbffe10fff 64bit]
[   72.068150] pci d4a8:00:00.0: BAR 2: assigned [mem 0xbffe11000-0xbffe11fff 64bit]
[   72.068380] pci d4a8:00:00.0: BAR 4: assigned [mem 0xbffe12000-0xbffe12fff 64bit]
[   72.171482] hv_pci 873a5444-894f-4737-97ba-d5d84c001ba2: PCI VMBus probing: Using version 0x10004
[   72.199268] hv_pci 873a5444-894f-4737-97ba-d5d84c001ba2: PCI host bridge to bus 894f:00
[   72.199270] pci_bus 894f:00: root bus resource [mem 0xbffe14000-0xbffe16fff window]
[   72.199271] pci_bus 894f:00: No busn resource found for root bus, will use [bus 00-ff]
[   72.199841] pci 894f:00:00.0: [1af4:1049] type 00 class 0x010000
[   72.200280] pci 894f:00:00.0: reg 0x10: [mem 0xbffe14000-0xbffe14fff 64bit]
[   72.200548] pci 894f:00:00.0: reg 0x18: [mem 0xbffe15000-0xbffe15fff 64bit]
[   72.201140] pci 894f:00:00.0: reg 0x20: [mem 0xbffe16000-0xbffe16fff 64bit]
[   72.203072] pci_bus 894f:00: busn_res: [bus 00-ff] end is updated to 00
[   72.203076] pci 894f:00:00.0: BAR 0: assigned [mem 0xbffe14000-0xbffe14fff 64bit]
[   72.203313] pci 894f:00:00.0: BAR 2: assigned [mem 0xbffe15000-0xbffe15fff 64bit]
[   72.203548] pci 894f:00:00.0: BAR 4: assigned [mem 0xbffe16000-0xbffe16fff 64bit]
```

And the output of syslog looks like this:

```
Dec 11 11:02:21 TDixon systemd[1]: Created slice User Slice of UID 1000.
Dec 11 11:02:21 TDixon systemd[1]: Starting User Runtime Directory /run/user/1000...
Dec 11 11:02:21 TDixon systemd[1]: Finished User Runtime Directory /run/user/1000.
Dec 11 11:02:21 TDixon systemd[1]: Starting User Manager for UID 1000...
Dec 11 11:02:21 TDixon systemd[1081]: Queued start job for default target Main User Target.
Dec 11 11:02:21 TDixon systemd[1081]: Created slice User Application Slice.
Dec 11 11:02:21 TDixon systemd[1081]: Reached target Paths.
Dec 11 11:02:21 TDixon systemd[1081]: Reached target Timers.
Dec 11 11:02:21 TDixon systemd[1081]: Listening on GnuPG network certificate management daemon.
Dec 11 11:02:21 TDixon systemd[1081]: Listening on GnuPG cryptographic agent and passphrase cache (access for web browsers).
Dec 11 11:02:21 TDixon systemd[1081]: Listening on GnuPG cryptographic agent and passphrase cache (restricted).
Dec 11 11:02:21 TDixon systemd[1081]: Listening on GnuPG cryptographic agent (ssh-agent emulation).
Dec 11 11:02:21 TDixon systemd[1081]: Listening on GnuPG cryptographic agent and passphrase cache.
Dec 11 11:02:21 TDixon systemd[1081]: Listening on debconf communication socket.
Dec 11 11:02:21 TDixon systemd[1081]: Listening on REST API socket for snapd user session agent.
Dec 11 11:02:21 TDixon systemd[1081]: Reached target Sockets.
Dec 11 11:02:21 TDixon systemd[1081]: Reached target Basic System.
Dec 11 11:02:21 TDixon systemd[1081]: Reached target Main User Target.
Dec 11 11:02:21 TDixon systemd[1]: Started User Manager for UID 1000.
Dec 11 11:02:21 TDixon systemd[1081]: Startup finished in 27ms.
Dec 11 11:02:21 TDixon systemd[1]: Started Session c7 of User tdixon.
Dec 11 11:02:21 TDixon systemd[1]: session-c7.scope: Deactivated successfully.
Dec 11 11:02:21 TDixon systemd[1]: Started Session c8 of User tdixon.
Dec 11 11:02:21 TDixon systemd[1]: session-c8.scope: Deactivated successfully.
Dec 11 11:02:22 TDixon systemd[1]: Started Session c9 of User tdixon.
Dec 11 11:02:22 TDixon systemd[1]: session-c9.scope: Deactivated successfully.
Dec 11 11:02:22 TDixon systemd[1]: Started Session c10 of User tdixon.
Dec 11 11:02:22 TDixon systemd[1]: session-c10.scope: Deactivated successfully.
Dec 11 11:02:22 TDixon systemd[1]: Started Session c11 of User tdixon.
Dec 11 11:02:22 TDixon systemd[1]: session-c11.scope: Deactivated successfully.
Dec 11 11:02:22 TDixon systemd[1]: Started Session c12 of User tdixon.
Dec 11 11:02:22 TDixon systemd[1]: session-c12.scope: Deactivated successfully.
```

dorssel, 2023-12-11:
Hmmm, the fact that there is absolutely nothing in the Linux log means that it is failing really early on.

Let me just verify: `usbipd attach --wsl` (4.0.0 syntax) doesn't work, even though it is using the correct distro and IP address. But manually running `usbip attach` from within WSL is working correctly?

Since 4.0.0 `usbipd attach --wsl` uses a built-in `usbip` client tool instead of any distro tools.

Could you try from within WSL, but instead of using `usbip` (the distro client tools) use `"/mnt/c/Program Files/usbipd-win/wsl/usbip"` as client tool? Does that work?

elcapitantommy, 2023-12-11:
> Let me just verify: `usbipd attach --wsl` (4.0.0 syntax) doesn't work, even though it is using the correct distro and IP address. But manually running `usbip attach` from within WSL is working correctly?

Correct.

> Could you try from within WSL, but instead of using `usbip` (the distro client tools) use `"/mnt/c/Program Files/usbipd-win/wsl/usbip"` as client tool? Does that work?

`"/mnt/c/Program Files/usbipd-win/wsl/usbip" attach -r 172.22.112.1 -b 2-1` works just fine.

dorssel, 2023-12-11:
That is *exactly* the command that is run by `usbipd-win`, but from the Windows side as:
```
wsl --distribution Ubuntu --user root -- "/mnt/c/Program Files/usbipd-win/wsl/usbip" attach -r 172.22.112.1 -b 2-1
```
If that works from the command line, then I am clueless why it doesn't work as `usbipd attach --wsl --busid 2-1`...

elcapitantommy, 2023-12-11:
It's very strange.

I opened this ticket originally because for some reason I was having stability issues when connecting from WSL instead of Windows but I have since upgraded to a powered USB hub which has resolved that issue.

Now (_just because I'm stubborn_) it's annoying that there's no apparent explanation for what's going on (but I appreciate your efforts to help)!

I'm happy for this issue to be closed if you wish, but if you want to keep it open I'm happy to assist with any further troubleshooting you may think of.

dorssel, 2023-12-11:
Power issues could explain flaky behavior. When you attach, the device is reset which causes a temporarily higher consumption. Under-power could cause the device to reset multiple times, which could lead to weird symptoms (not attaching at all, attaching but immediately detaching again, etc. etc.).

I'll close this, as you seem to have resolved your issue, and I don't see resemblance with reports from other users.

closed 2023-12-11 by dorssel

Rjgawuie, 2024-12-29:
I got a same problem recently, I solved it through add sudo front of command.