dorssel/usbipd-win #771: update-alternatives: warning: forcing reinstallation of alternative /usr/lib/linux-tools/5.15.0-88-generic/usbip because link group usbip is broken update-alternatives: error: error creating symbolic link '/usr/local/bin/usbip.dpkg-tmp': No such file or directory

AdeebIsmail, 2023-11-18:
sudo update-alternatives --install /usr/local/bin/usbip usbip `ls /usr/lib/linux-tools/*/usbip | tail -n1` 20

running this command gives me the error 

update-alternatives: warning: forcing reinstallation of alternative /usr/lib/linux-tools/5.15.0-88-generic/usbip because link group usbip is broken update-alternatives: error: error creating symbolic link '/usr/local/bin/usbip.dpkg-tmp': No such file or directory

dorssel, 2023-11-18:
What distro?

AdeebIsmail, 2023-11-18:
WSL Ubuntu

dorssel, 2023-11-18:
I see no reason why it would even try to access `/usr/local/bin/`. Maybe you should repair first, using `sudo apt install -f`.

AdeebIsmail, 2023-11-18:
Tried that, I still get the same error

dorssel, 2023-11-18:
`/usr/local/bin/usbip...` indicates that somehow you have (tried to?) install a package manually. And now your package system is confused. The default Ubuntu packages don't use `/usr/local`.

snowgato, 2023-11-30:
Same issue for me with Ubuntu22.04, no idea how to fix it ?

dorssel, 2023-11-30:
What is the output of:
```
ls -la /usr/local/bin/usbip
ls -la /usr/bin/usbip
ls -la /usr/lib/linux-tools
ls -la /usr/lib/linux-tools/*/usbip
```

snowgato, 2023-11-30:
Hi,
`cedric@kakak:~$ ls -la /usr/bin/usbip
-rwxr-xr-x 1 root root 1622 Oct 30 21:15 /usr/bin/usbip

cedric@kakak:~$ ls -la /usr/bin/usbip
-rwxr-xr-x 1 root root 1622 Oct 30 21:15 /usr/bin/usbip

cedric@kakak:~$ ls -la /usr/lib/linux-tools
total 12
drwxr-xr-x  3 root root 4096 Nov 30 12:04 .
drwxr-xr-x 75 root root 4096 Nov 30 12:04 ..
drwxr-xr-x  2 root root 4096 Nov 30 12:04 5.15.0-89-generic

cedric@kakak:~$ ls -la /usr/lib/linux-tools/*/usbip
lrwxrwxrwx 1 root root 33 Oct 30 21:15 /usr/lib/linux-tools/5.15.0-89-generic/usbip -> ../../linux-tools-5.15.0-89/usbip
`
Thanks for your help !

dorssel, 2023-11-30:
You missed the first one (the one that the error is actually complaining about):
```
ls -la /usr/local/bin/usbip
```

and one more, for good measure:

```
ls -la /usr/lib/*/usbip
```

snowgato, 2023-11-30:
Oh sorry,
Here it is:

root@kakak:/home/cedric# ls -la /usr/local/bin/usbip
-rw-r--r-- 1 root root 0 Oct 15 23:44 /usr/local/bin/usbip

and 

root@kakak:/home/cedric# ls -la /usr/lib/*/usbip
-rwxr-xr-x 1 root root 248920 Oct 30 21:15 /usr/lib/linux-tools-5.15.0-89/usbip

dorssel, 2023-11-30:
> -rw-r--r-- 1 root root 0 Oct 15 23:44 /usr/local/bin/usbip

This one is the problem. This is supposed to be a symlink, but it is instead an empty file. Just remove it and run `update-alternatives` again. That should fix it.

snowgato, 2023-11-30:
Perfect, thanks !
I can see the device with lsusb command but not with lsblk
Is there a way to get it listed with lsblk and then mount it once appearing on lsusb output command ?

dorssel, 2023-11-30:
Search through the other issues. WSL kernel by default does not have USB storage drivers. You need to compile your own, see https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-usbip-enabled-wsl-2-kernel.

snowgato, 2023-12-01:
Thanks for the link. Actually all the recommended kernel features where already builtin the kernel. I have added USB mass storage support as adviced on this [tutorial](https://github.com/rpasek/usbip-wsl2-instructions/blob/master/README.md) . I made a mix of your tutorial and the other one.
And it works !!
I am just begininng to use it, anyway I've been stuck with usb device being busy and noway other than rebooting to get it non busy. Each time I disconnected it and reconnected it, it was occupied by some process and didn't succeed to find it ( using resmon or processExplorer ). I finally rebooted , usb drive disconnected and have been able to connect it to wsl.
I am bit afraid of getting stuck other times with devices getting busy without being able to identify the occupying process.

dorssel, 2023-12-01:
If you don't need to use the device on Windows at all, I recommend using `usbipd bind --force`

dorssel, 2023-12-02:
That is correct.

dorssel, 2023-12-06:
4.0.0 has been released and installing client-side tooling is no longer necessary.