dorssel/usbipd-win #773: usbipd state vs list speed

andrewleech, 2023-11-22:
I make heavy use of `usbipd state` in my WSL USB Manager GUI and it works great, certainly far more reliable than parsing the human readable `list` command so thanks for adding it! 

There's been a number of times recently I've noticed however that running this can be significantly slower than `usbipd list`. 

I presume this is due to some of the extra details included in `state`?

It's not _too_ bad on my computer currently, sometimes it's noticible slower, but there's still a fair difference:
``` powershell
PS C:\> Measure-Command -Expression { usbipd list } | select TotalSeconds

TotalSeconds
------------
   0.3065414
```
``` powershell
PS C:\> Measure-Command -Expression { usbipd state } | select TotalSeconds

TotalSeconds
------------
   1.3726969
```

Do you know what might be causing the delay? Are there some fields that require querying wsl / usbip client that would be contributing to this? 
Would it be straightforward to add an option like `usbipd state --quick` that skips these fields (assuming I don't need them😉)?

dorssel, 2023-11-22:
Yes, I am aware of this. And this has already been resolved in current master. Mostly due to the following: https://github.com/dorssel/usbipd-win/discussions/774. Please have a look at that, it will affect your project.

andrewleech, 2023-11-22:
Thanks for your great work!

closed 2023-11-22 by andrewleech