dorssel/usbipd-win #791: usb storage device doesn't show up anymore in windows after unbind

snowgato, 2023-12-07:
Hi,
I am having trouble unbinding a usb ssd drive running ubuntu-22.04 on wsl2.
the usb drive is listed by usbipd wsl list command as not attached, then I unbind it.
There's a bell sound, but it doesn't appear on windows explorer and nor on device manager under disks section, even if I reboot the laptop.
Using a linux distribution, this usb drive works fine but I can't see it anymore on windows 11.
Thanks very much for your help !

dorssel, 2023-12-07:
What is the current output of `usbipd list`?

snowgato, 2023-12-07:
3-1    0781:558c  Extreme SSD                                                   Not attached

dorssel, 2023-12-07:
That is output from `usbipd wsl list`, what does `usbipd list` say?

snowgato, 2023-12-07:
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-2    046d:c52b  Logitech USB Input Device, USB Input Device                   Not shared
2-7    04f2:b6f9  Chicony USB2.0 Camera, IR Camera                              Not shared
2-10   8087:0032  Intel(R) Wireless Bluetooth(R)                                Not shared
3-1    0781:558c  Extreme SSD                                                   Not shared
3-2    0781:55b1  USB Mass Storage Device                                       Shared
3-4    1058:2620  USB Mass Storage Device                                       Not shared

Persisted:
GUID                                  DEVICE
789294d6-e3e5-492a-80f6-12f8f547c4b8  USB Mass Storage Device
9a1f9345-efe6-4a7e-a8ca-0ff2e3e024fb  USB Mass Storage Device

dorssel, 2023-12-07:
When you previously shared the SSD with WSL, did you (ever) use `usbipd bind --force`?

snowgato, 2023-12-07:
yes I did

dorssel, 2023-12-07:
That could explain it. When you use `--force`, usbipd forcefully alters the driver for the device to the USB-stub driver required for forwarding. When you then `unbind` the device, it removes that driver and tells Windows to install whatever Windows thinks should be the driver (default). Does this device require special drivers from the manufacturer? Is there an installer for that?

You can check in device manager what driver (if any) is currently associated with the device. It should in any case not be the USBIP stub driver (usbipd should have removed that when you used `unbind`).

snowgato, 2023-12-07:
I haven't explictly installed any driver and have never looked before if there was a special driver installed for it. Currently I don't see the device in disk devices section of device manager

dorssel, 2023-12-07:
If you want, can you test one more time:
- `usbipd bind --force`
- `usbipd unbind`
The intention is that the default driver (which you now installed manually using "update driver") is actually restored correctly by unbind itself. If it isn't, then that is a bug. But maybe it was a one-off fluke the first time...

llahteinen, 2025-03-11:
I think I faced this bug.
usbipd version 4.4.0-128+Branch.master.Sha.d52ae5f18f93deb85cd4e1722ca5f812443a2c6d.d52ae5f18f93deb85cd4e1722ca5f812443a2c6d

I used this guide to attach a joystick to wsl
https://learn.microsoft.com/en-us/windows/wsl/connect-usb

usbipd bind failed and advised to try --force, so I did. After that I rebooted the host and continued with attach. Then I was able to see the device inside WSL with lsusb.

After I was done, I did usbipd detach --busid <busid>
But my joystick didn't reappear on the host side.
So I tried usbipd unbind. This did not make the joystick reappear on the host.

The state was "Not shared", but I noticed that the name was now "USBIP Shared Device" and not "Logitech xyz".

I was able to finally get the joystick working on the host by running:
```
Get-PnpDevice -PresentOnly | Where-Object { $_.InstanceId -match '^USB' } | SELECT Name,InstanceId
pnputil /remove-device "<InstanceId>"
```
And the reattaching the joystick.
Which took quite a while to figure out, I haven't needed to use those commands in my 30 years of using Windows.

dorssel, 2025-03-11:
@llahteinen 
This is a known regression bug in 4.4.0. See https://github.com/dorssel/usbipd-win/issues/1160#issuecomment-2709477268