dorssel/usbipd-win #794: Unable to attach using busid or hardware id

bbtaylr, 2023-12-10:
Installed usbipd 4.0.0 to enable USB device within Docker linux container running on Windows 10 with WSL - 5.15.133.1-microsoft-standard-WSL2 kernel

Issued the following at cmd prompt and saw error and attach was unsuccessful:

usbipd attach --wsl docker-desktop --busid 1-4
usbipd: info: Selecting a specific distribution is no longer required. Please file an issue if you believe that the default selection mechanism is not working for you.
usbipd: info: Using WSL distribution 'docker-desktop' to attach; the device will be available in all WSL 2 distributions.

usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.

dorssel, 2023-12-10:
What is the output of `wsl --list --all --verbose`?

bbtaylr, 2023-12-10:
the output of wsl --list --all --verbose is as follows

NAME                   STATE           VERSION
* docker-desktop         Running         2
  Ubuntu                 Stopped         2
  docker-desktop-data    Running         2

dorssel, 2023-12-11:
What is the output of
```
wsl --distribution docker-desktop --user root -- mount
wsl --distribution docker-desktop --user root -- ls -l '/mnt/c/Program Files/usbipd-win/wsl'
```

Does it work correctly if you start the `Ubuntu` distribution and attach using `--distribution Ubuntu`?

eshikafe, 2024-01-14:
I am having the same problem

```
PS C:\Users\austin> wsl --version
WSL version: 2.0.9.0
Kernel version: 5.15.133.1-1
WSLg version: 1.0.59
MSRDC version: 1.2.4677
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.19045.3803
```

PS C:\Users\austin> usbipd attach -b 1-2 -w
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
**usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.**
PS C:\Users\austin>

```
PS C:\Users\austin>  wsl --list --all --verbose
  NAME      STATE           VERSION
* Ubuntu    Running         2

PS C:\Users\austin> usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
1-1    0000:0004  Unknown USB Device (Set Address Failed)                       Not shared
1-2    30c9:000f  HP HD Camera, HP IR Camera                                    Shared
1-4    046d:c534  USB Input Device                                              Not shared
1-7    06cb:00f0  Synaptics FS7605 Touch Fingerprint Sensor with PurePrint(TM)  Not shared
1-10   8087:0026  Intel(R) Wireless Bluetooth(R)                                Not shared
3-1    8087:0af1  HP GNSS Sensor                                                Not shared
3-2    8087:0aca  USB Serial Device (COM3), USB Serial Device (COM4), Modem...  Not shared
3-3    8087:0ac9  Intel(R) XMM(TM) 7360 LTE-A                                   Not shared

Persisted:
GUID                                  DEVICE

usbipd: warning: Unknown USB filter 'CsDeviceControl' may be incompatible with this software; 'bind --force' may be required.
PS C:\Users\austin>
```

eshikafe, 2024-01-14:
PS C:\Users\austin> usbipd attach -i 30c9:000f -w
**usbipd: info: Device with hardware-id '30c9:000f' found at busid '1-2'.**
**usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.**
**usbipd: error: Unable to run 'usbip' client tool. Please report this at https://github.com/dorssel/usbipd-win/issues.**
PS C:\Users\austin>

dorssel, 2024-01-14:
Please try the latest master build instead of the release. It has a fix that should support docker-desktop better. You can find it here:  https://github.com/dorssel/usbipd-win/actions/runs/7472527220

eshikafe, 2024-01-15:
I am not using Docker. I have wsl installed directly on Windows. Thanks.

eshikafe, 2024-01-15:
After upgrading wsl from 2.0.9 to 2.0.14, I get a warning message that "A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed"

```
PS C:\Users\austin> wsl --update
Checking for updates.
Updating Windows Subsystem for Linux to version: 2.0.14.

PS C:\Users\austin> wsl --version
WSL version: 2.0.14.0
Kernel version: 5.15.133.1-1
WSLg version: 1.0.59
MSRDC version: 1.2.4677
Direct3D version: 1.611.1-81528511
DXCore version: 10.0.25131.1002-220531-1700.rs-onecore-base2-hyp
Windows version: 10.0.19045.3803 
```

```
PS C:\Users\austin> usbipd attach -b 1-2 -w
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address x.x.x.x to reach the host.
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
```

dorssel, 2024-01-15:
> usbipd: info: Using IP address x.x.x.x to reach the host.

Considering the warning is about networking/firewalling, you've hided some critical information required to debug this problem...

eshikafe, 2024-01-15:
Yes. I removed the IP. Here is the complete message

```
PS C:\Users\austin> usbipd attach -b 1-2 -w
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.24.48.1 to reach the host.
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
```
```
C:\Users\austin>ipconfig

Windows IP Configuration

Ethernet adapter vEthernet (WSL):

   Connection-specific DNS Suffix  . :
   Link-local IPv6 Address . . . . . : fe80::351d:57c6:5203:8f25%70
   IPv4 Address. . . . . . . . . . . : 172.24.48.1
   Subnet Mask . . . . . . . . . . . : 255.255.240.0
   Default Gateway . . . . . . . . . :
```

dorssel, 2024-01-15:
@eshikafe 
Is the actual attach working? Or does it "hang" trying to connect?

eshikafe, 2024-01-16:
No, it isn't working

```
PS C:\Users\austin> usbipd attach -w ubuntu -b 2-2
usbipd: info: Selecting a specific distribution is no longer required. Please file an issue if you believe that the default selection mechanism is not working for you.
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.24.48.1 to reach the host.
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '2-2'.
```

dorssel, 2024-01-17:
@eshikafe 
Are you perhaps using Windows 11 with WSL in mirrored networking mode?
Are you using any 3rd party security software?
Can you `ping` the Windows host at 172.24.48.1 from WSL?

dorssel, 2024-01-17:
@eshikafe 
OK, next guess. Maybe the firewall rule wasn't installed correctly, or has been disabled somehow. In the advanced Windows Firewall settings there should be an (enabled) entry `usbipd`. You could try a repair install or reinstall of `usbipd-win` to see if that fixes it.

yanbingyu07, 2024-01-24:
@dorssel, I got the exact same problem as @eshikafe described. I went through this thread from beginning, I checked the firewall rule, which is correctly installed; I even reinstalled `usbipd-win`, but not fixed. Any further ideas to check? Thanks!

dorssel, 2024-01-24:
@yanbingyu07 
Just to confirm: you can ping your Windows host at the listed address. And the `usbipd` service is running. And you are *not* using Windows 11 with WSL in mirrored networking mode. Right?

The firewall rule allows all local networks to connect. Maybe (just maybe) Windows is confused and does not regard the incoming connection from WSL as "local network". Then it would be blocked. Maybe you can try with the firewall turned off (carefully, of course, turn it back on right after the test). Or maybe change the firewall rule so that all incoming connections are allowed, not just local networks.

I'm just guessing, though. I don't really get why it isn't working for you.

yanbingyu07, 2024-01-24:
@dorssel, Thanks for your reply!
> Just to confirm: you can ping your Windows host at the listed address. And the usbipd service is running. And you are not using Windows 11 with WSL in mirrored networking mode. Right?

Yes to all above, but host OS is Windows 10 Pro.

> The firewall rule allows all local networks to connect. Maybe (just maybe) Windows is confused and does not regard the incoming connection from WSL as "local network". Then it would be blocked. [...]

The firewall rule for `usbipd` applied to all profiles: Domain/Private/Public, as shown below. I will play more with firewall settings.

[image] 
[image]

yanbingyu07, 2024-01-24:
@dorssel, 
BTW, I can use the ssh tunnel/reverse proxy method you mentioned in [other post](https://github.com/dorssel/usbipd-win/issues/335#issuecomment-1221389478) to get around. 

This tcp port error happened just now, it was working fine two months ago. I am curious about the cause of this. It could be changes in corp network that I am not aware of.

marbaa, 2024-02-27:
Is it working for anybody at all?

```
> usbipd.exe attach -w -b 7-4
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.29.112.1 to reach the host.
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '7-4'.
```

Firewall rule is in place.
[image]
[image]
[image]

dorssel, 2024-02-27:
> Is it working for anybody at all?

For about 10.000+ people, yes.

> usbipd: warning: A *third-party* firewall

Are you using any other security software? Are you domain-joined and are company policies in effect? Can you ping to the listed IP address from WSL2 to Windows?

marbaa, 2024-02-27:
> > Is it working for anybody at all?
> 
> For about 10.000+ people, yes.
> 

That's very cool  🙂 I'm trying to make it work for half a day, including reading many, many github topics and couldn't find any solution to it.

I'm testing it on domain-joined corporate notebook. We are using Microsoft Defender only. I can ping listed IP. I tried to update WSL from stable v2.0.4 (I guess) to v2.1.3, just in case. I tried ssh -R tunnel and mount usb dongle from wsl debian but got error message 

```
usbip attach -b 7-4 -r 127.0.0.1
usbip: error: import device
```

I have admin rights, I can create/remove firewall rules in "Inbound" section, but I can't make any change in "Allow an app through firewall" executed from Defender. All entries are grayed for me, and I'm not sure if the list is actual, realtime output if it is grayed, because currently, usbipd is not checked in the list.

[image]

dorssel, 2024-02-27:
I too am on a corporate domain, and here all `usbipd` boxes are checked. I think this is the problem; direct connection is not going to work due to company policy constraints. However, the `ssh` tunnel should work; other people have solved their connectivity problems this way.

marbaa, 2024-02-28:
For me ssh tunnel doesn't work. Probably my wifi dongle is not well supported in Linux as I found out in Linux VM running in VMware player. I wanted to compile new driver according [this nice repo](https://github.com/clnhub/rtl8192eu-linux) but I gave up. In Linux VM it was easy, but in WSL there is missing ```linux-headers-5.15.146.1-microsoft-standard-WSL2``` package. Found another steps how to get this package, but thats too much work to be done, just for testing wifi dongle in wsl. 😟

I need to bypass our vpn for working, and currently using VMware Player VM with passthrough wifi dongle. I wanted to get rid of this VM and have wifi dongle connected to wsl and redirect traffic with ssh into wsl.

dorssel, 2024-02-28:
@marbaa
> For me ssh tunnel doesn't work.

Why not? What is the error/log?

> In Linux VM it was easy, but in WSL there is missing linux-headers-5.15.146.1-microsoft-standard-WSL2 package.

Compiling a WSL kernel is explained here: https://github.com/dorssel/usbipd-win/wiki/WSL-support#building-your-own-wsl-2-kernel-with-additional-drivers

marbaa, 2024-03-06:
> @marbaa
> 
> > For me ssh tunnel doesn't work.
> 
> Why not? What is the error/log?
> 

```
usbip attach -b 7-4 -r 127.0.0.1
usbip: error: import device
```

Thanks for link, I briefly tried to compile new kernel, but it wasn't as straightforward as it looked with docu. I was getting many errors, so I gave up again.

I don't have much time to spend on this 😟Maybe I will come back to this in few months, or when I get a new wifi dongle with better Linux support. I will keep working with wifi dongle pass-though to vmware player.

dorssel, 2024-03-10:
Original problem (not finding `usbip` client tools) was fixed in 4.1.0.

closed 2024-03-10 by dorssel

marbaa, 2024-03-16:
Just note, after long searching and testing, I found this [issue](https://github.com/microsoft/WSL/issues/4139) in WSL repo. Defender is blocking traffic to WSL, because it treats vEthernet as public network, see [comment](https://github.com/microsoft/WSL/issues/4139#issuecomment-1028835084) I unchecked vEthernet (WSL) from mentioned picture in that issue and then the firewall message disappeared. However as it is corporate notebook, I can't leave it unchecked, Defender got red immediately, that it is not compliant.

dorssel, 2024-03-17:
@marbaa 
That is an excellent find!

I was able to reproduce it like this. And I also made a detection for it in #895.
Could you please test the installer from https://github.com/dorssel/usbipd-win/actions/runs/8317239301?pr=895 and see if it correctly detects and diagnoses the issue in your situation?

Thanks for the report!

marbaa, 2024-03-18:
I uninstalled 4.1.0 and  installed 4.1.1. I don't see any different output and haven't noticed any additional behaviour.

```
C:\Users\user>usbipd --version
4.1.1-PullRequest0895.28+Branch.pull-895-merge.Sha.6cd7c8e5622d706ecbb0c57c1cf0e5b3fa493b51

C:\Users\user>
```

```
C:\Users\user>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-3    1bcf:28c4  Integrated Webcam                                             Not shared
2-5    0a5c:5843  Dell ControlVault w/ Fingerprint Touch Sensor, Microsoft ...  Not shared
2-11   2357:0109  TP-Link Wireless USB Adapter                                  Not shared
```

```
C:\Users\user>usbipd bind -b 2-11

C:\Users\user>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-3    1bcf:28c4  Integrated Webcam                                             Not shared
2-5    0a5c:5843  Dell ControlVault w/ Fingerprint Touch Sensor, Microsoft ...  Not shared
2-11   2357:0109  TP-Link Wireless USB Adapter                                  Shared
```

```
C:\Users\user>usbipd attach --wsl -b 2-11
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.29.112.1 to reach the host.
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '2-11'.

C:\Users\user>
```
[image]

dorssel, 2024-03-18:
Ah. What I detect (and this is the only way how I get the error) is when the setting for Inbound connections is "Always block", not "Block (Default)". That is now detected:

```
usbipd: info: Using WSL distribution 'wsl-usbip' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.28.96.1 to reach the host.
usbipd: warning: Windows firewall is blocking all incoming connections for public interfaces (including WSL).
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.
```

dorssel, 2024-03-18:
@marbaa 
Can you check whether local firewall rules are applied for the public profile, as shown here:

https://github.com/microsoft/WSL/issues/4139#issuecomment-766316939 and https://github.com/microsoft/WSL/issues/4139#issuecomment-850356894

(PS, the case I now check for is this one: https://github.com/microsoft/WSL/issues/4139#issuecomment-1200345491

marbaa, 2024-03-18:
@dorssel 
[image]

Not sure if I want to try change "Apply local firewall rules" on corporate notebook 🙂I'm afraid of security team. 😁

dorssel, 2024-03-18:
OK, so that is it in your case: the firewall "allow rule" is not applied to public networks (which unfortunately is the profile the WSL switch is under). I will add a check for this one too, so an appropriate warning is displayed for it.

dorssel, 2024-03-19:
@marbaa 
Can you give the new PR build a try? It is at https://github.com/dorssel/usbipd-win/actions/runs/8336986169?pr=895.
(you will probably have to uninstall/reinstall usbipd-win, since the version numbers are still the same).

marbaa, 2024-03-20:
I've uninstalled and installed new version. Output is not changed I guess:

```
C:\Users\user>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-3    1bcf:28c4  Integrated Webcam                                             Not shared
2-5    0a5c:5843  Dell ControlVault w/ Fingerprint Touch Sensor, Microsoft ...  Not shared
2-11   1ea7:0066  USB Input Device                                              Not shared
2-13   2357:0109  TP-Link Wireless USB Adapter                                  Shared
2-14   8087:0032  Intel(R) Wireless Bluetooth(R)                                Not shared
3-4    0bda:8153  Realtek USB GbE Family Controller                             Not shared
6-1    0bda:5400  BillBoard Device                                              Not shared
6-5    413c:b06e  USB Input Device                                              Not shared
7-1    0b0e:0343  Jabra UC VOICE 150a, USB Input Device                         Not shared
7-3    045e:076d  Microsoft® LifeCam HD-5000                                    Not shared
7-5    413c:b06f  USB Input Device                                              Not shared
8-3    0b05:1877  USB Input Device                                              Not shared
8-4    0bf8:1028  USB Input Device                                              Not shared

Persisted:
GUID                                  DEVICE

C:\Users\user>usbipd attach -b 2-13 --wsl
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.29.112.1 to reach the host.
usbipd: warning: A firewall may be blocking the connection; ensure TCP port 3240 is allowed.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '2-13'.

C:\Users\user>
```

Other thing - regarding **ssh tunnel**.
Before, I was getting ```usbip: error: import device``` when attaching dongle through ssh tunnel. I attached it with sudo, and it went fine. However I don't see the interface, I guess, thats because of missing driver/default WSL kernel not supporting kernel modules?

```
user@wsl:~$ usbip list -r 127.0.0.1
Exportable USB devices
======================
 - 127.0.0.1
       2-13: TP-Link : TL-WN823N v2/v3 [Realtek RTL8192EU] (2357:0109)
           : USB\VID_0E0F&PID_0001\00E04C000001
           : (Defined at Interface level) (00/00/00)
           :  0 - Vendor Specific Class / Vendor Specific Subclass / Vendor Specific Protocol (ff/ff/ff)
```
```
user@wsl:~$ sudo usbip attach --remote=127.0.0.1 --busid=2-13
user@wsl:~$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1310 qdisc mq state UP group default qlen 1000
    link/ether 00:15:5d:33:af:ec brd ff:ff:ff:ff:ff:ff
    inet 172.29.125.110/20 brd 172.29.127.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::215:5dff:fe33:afec/64 scope link
       valid_lft forever preferred_lft forever
```
```
user@wsl:~$ dmesg
...
[ 3805.343912] usb 1-1: new high-speed USB device number 2 using vhci_hcd
[ 3805.493929] usb 1-1: SetAddress Request (2) to port 0
[ 3805.529530] usb 1-1: New USB device found, idVendor=2357, idProduct=0109, bcdDevice= 2.00
[ 3805.529923] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3805.530208] usb 1-1: Product: 802.11n NIC
[ 3805.530369] usb 1-1: Manufacturer: Realtek
[ 3805.530502] usb 1-1: SerialNumber: 00e04c000001
user@wsl:~$
user@wsl:~$ sudo lsmod
Module                  Size  Used by
user@wsl:~$
```

dorssel, 2024-03-20:
@marbaa 
Hmmm, I really wonder how your connections are blocked. I check for both Local Admin settings and group policies (Domain Admin) now. Can you share what is in your registry (regedit.exe) at

```
SYSTEM\CurrentControlSet\Services\SharedAccess\Parameters\FirewallPolicy\PublicProfile
```

and

```
SOFTWARE\Policies\Microsoft\WindowsFirewall\PublicProfile
```

And yes, it's missing drivers. Default kernel does indeed not support loadable modules, so you will have to compile your own and statically link in the needed drivers. USB network (including WiFi) is not enabled in stock WSL at all. Other people have made it work, however; search through older issues/discussions.

marbaa, 2024-03-21:
> @marbaa Hmmm, I really wonder how your connections are blocked. I check for both Local Admin settings and group policies (Domain Admin) now. Can you share what is in your registry (regedit.exe) at
> 
> ```
> SYSTEM\CurrentControlSet\Services\SharedAccess\Parameters\FirewallPolicy\PublicProfile
> ```
[image]

> and
> 
> ```
> SOFTWARE\Policies\Microsoft\WindowsFirewall\PublicProfile
> ```
I don't have 'WindowsFirewall' in whole registry. I search for PublicProfile and it gave me multiple results.

dorssel, 2024-03-21:
I really wonder where/how your Windows Firewall gets its "Apply local firewall rules: No" configuration. Would be nice if you somehow can figure that out. The only place I know of is the group policy setting, which I test for here:

https://github.com/dorssel/usbipd-win/blob/windows_firewall/Usbipd/Wsl.cs#L45-L48

returntrip, 2024-07-09:
> Just note, after long searching and testing, I found this [issue](https://github.com/microsoft/WSL/issues/4139) in WSL repo. Defender is blocking traffic to WSL, because it treats vEthernet as [...]

Thanks! This really helped me a lot. But I do have the same problem as you, Defender marking me as not compliant.

A couple of lines to make the process a bit faster:

`Set-NetFirewallProfile -Profile "Public" -DisabledInterfaceAliases "vEthernet (WSL (Hyper-V firewall))"` to disable the interface

`Set-NetFirewallProfile -Profile "Public" -DisabledInterfaceAliases @()` to re enable the interface and pacify Defender :)

Johnny8923, 2024-07-26:
Hi guys,
Thanks for this thread, I ran into the same issue and one solution which worked for me was to mirror my network configuration to not be annoyed anymore by the Hyper-V firewall. To do this, I've added the option: `networkingMode=mirrored` to my .wslconfig file and restarted wsl.

addshore, 2024-07-29:
> To do this, I've added the option: networkingMode=mirrored to my .wslconfig file and restarted wsl.

> With this solution Defender will not complain.

+1 to this, I switch from `networkingMode=NAT` to `networkingMode=mirrored` and this instantly started working.

However, then a bunch of other stuff for me breaks, such as accessing my docker containers via localhost from windows land :)

returntrip, 2024-09-30:
> firewall a little, and flicked `edge traversal` from `block edge traversal` to `allow edge traversal` and then things seem to connect as expected.
> 
> I have actually since flicked this setting back to block now, and restarted WSL etc and it all appears to work???
> 
> Just leaving this here incase it works for the next person, no idea what has happened under the hood, and maybe it was something else I clicked too?

This did not work for me, but i wonder, have you changed anything else on this FW rule?

marbaa, 2024-11-30:
I built WSL kernel with enabled modules, so I was able to load kernel driver for my wifi adapter.

In my case, ```edge traversal``` from ```block edge traversal``` to ```allow edge traversal``` had no effect.

Interesting is, that with ```networkingMode=mirrored``` it works and firewall is not complaining anything:

```
PS C:\Users\xxx> usbipd.exe attach --wsl --busid 2-11
usbipd: info: Using WSL distribution 'Debian' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 127.0.0.1 to reach the host.
PS C:\Users\xxx>
```

Debian WSL2
```
16: wlx242fd06649ab: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP group default qlen 1000
    link/ether 24:2f:d0:66:49:ab brd ff:ff:ff:ff:ff:ff
    inet 192.168.222.125/24 brd 192.168.222.255 scope global dynamic wlx242fd06649ab
       valid_lft 3503sec preferred_lft 3503sec
```

However it is not usable for me as mirrored is corrupting other things, like @addshore mentioned.

@dorssel I couldn't find any difference in Windows Firewall settings when running WSL2 as NAT and as mirrored. Seems it is something inside WSL network infrastructure.

dorssel, 2024-12-01:
@returntrip 

> OK This is the only way I could workaround it without having Defender flagging me as uncompliant: `Set-ItemProperty -Path "HKLM:\Software\Policies\Microsoft\WindowsFirewall\PublicProfile" -Name AllowLocalPolicyMerge -Value 1` This allows merging of local FW rules

This is actually what usbipd-win (tries to) test for here: https://github.com/dorssel/usbipd-win/blob/cf18433f684eaf7c6594e1b458648ce0e9f47040/Usbipd/Wsl.cs#L46

Are you saying you did not get that warning? Maybe if this is not defined at all (neither 0 or 1) it is treated as 0?

returntrip, 2024-12-01:
> @returntrip
>
>
> This is actually what usbipd-win (tries to) test for here:
>
> https://github.com/dorssel/usbipd-win/blob/cf18433f684eaf7c6594e1b458648ce0e9f47040/Usbipd/Wsl.cs#L46
>
> Are you saying you [...]

I am quite sure I never got that warning but was always getting this: https://github.com/dorssel/usbipd-win/blob/cf18433f684eaf7c6594e1b458648ce0e9f47040/Usbipd/Wsl.cs#L555

dorssel, 2024-12-01:
> P.S.: and yes, that regkey was not defined at all for my system. So you might be completely right.

I'll try to find out what exactly the default is for this setting. If the default (non-existing) is always 0, then I will update the test.

addshore, 2024-12-12:
> > firewall a little, and flicked `edge traversal` from `block edge traversal` to `allow edge traversal` and then things seem to connect as expected.
> This did not work for me, but i wonder, have you changed anything else on this FW rule?

I did not change anything else on the firewall.
Maybe something else happened in the background, though?

Interestingly, `HKLM:\Software\Policies\Microsoft\WindowsFirewall\PublicProfile` is not defined on my system

Another interesting observation that I havn't been able to diagnose, sometimes on a fresh restart of my windows machine + WSL, usbipd will start spitting out the same or similar errors to this thread.
However, my solution to make them stop is simply restart again...
(I'll try to remember to make more notes and screenshots the next time this happens)

asyarifstudio, 2024-12-23:
I had this problem recently and it seems the problem comes when my VPN is active. Disabling it solve the issue

addshore, 2025-01-06:
**TLDR; your device might end up switching between a whole bunch of different networks, and interfaces, that windows may record and track seperatly, and out of the box, things will only work on `private` networks.**
**(If you are on windows 11 on a wired connection, you can use `Start > Run > ms-settings:network-ethernet` to get you to the right place for wired connections**

---

I moved my laptop from one location to another, hibernating in between, this also means a new external network.
Prior to moving `usbipd attach --busid 1-7 --wsl -a` had worked just fine, after moving it failed....

```
usbipd attach --busid 1-7 --wsl -a
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.21.224.1 to reach the host.
usbipd: warning: A firewall may be blocking the connection; ensure TCP port 3240 is allowed.
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.
```

I thought I would do some digging as I ended up in this state again...

1) I did a `wsl --shutdown`, restarted WSL, and this didn't achieve anything.
2) I disabled my tailscale VPN, as @asyarifstudio found turning off a VPN worked, but this didnt help me

3) I went ahead and disabled my Ethernet network connection in `Control Panel -> Network and Internet -> Network Connections` and re running the attach succeeded.

```
usbipd attach --busid 1-7 --wsl -a
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.21.224.1 to reach the host.
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.
WSL Attached
```

Note: This is technically a different ethernet adapter than I was using earlier (I think) as this is via a different USBC dock (my laptop has no built in ethernet).
So perhaps that is what has changed from an hour ago?
This also means that I don't encounter this issue while connected to my WiFi network it seems.

I then found that via this ethernet interface, and on this network, windows decided this was a public network

[image]

I'm not even sure where windows is getting `NETGEAR 2` from, as there isn't a single netgear device on this network....

This matches what I was seeing in network connections

[image]

Looking in `Network & internet > Ethernet` I see it as public again

[image]

Switching to private makes isbipd work again

```
usbipd attach --busid 1-7 --wsl -a
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.21.224.1 to reach the host.
usbipd: info: Starting endless attach loop; press Ctrl+C to quit.
WSL Attached
```