dorssel/usbipd-win #795: WLS2 cannot bind after setting the mirrored networking mode

wm692202103, 2023-12-11:
C:\Users\wm692\Desktop>usbipd list
Connected:
BUSID  VID:PID    DEVICE                                                        STATE
2-2    10c4:ea60  CP2102 USB to UART Bridge Controller                          Shared
2-3    8087:0032  英特尔(R) 无线 Bluetooth(R)                                   Not shared
2-4    13d3:56fb  Integrated Camera, Integrated IR Camera, Camera DFU Device    Not shared
3-2    0639:1000  QFMouse, USB 输入设备                                         Not shared
3-4    05ac:024f  USB 输入设备                                                  Not shared

Persisted:
GUID                                  DEVICE

C:\Users\wm692\Desktop>usbipd attach --wsl --busid 2-2
usbipd: info: Using WSL distribution 'Ubuntu-20.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 127.0.0.1 to reach the host.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '2-2'.

dorssel, 2023-12-11:
I cannot explain this, since there is a connectivity check to the service *before* the `usbip` connect.

What is the output of:
```
wsl --distribution Ubuntu-20.04 --user root -- bash -c 'echo < /dev/tcp/127.0.0.1/3240'
```

wm692202103, 2023-12-11:
> I cannot explain this, since there is a connectivity check to the service _before_ the `usbip` connect.
> 
> What is the output of:
> 
> ```
> wsl --distribution Ubuntu-20.04 --user root -- bash -c 'echo < /dev/tcp/127.0.0.1/3240'
> ```

C:\Users\wm692\Desktop>usbipd attach --wsl --busid 2-2
usbipd: info: Using WSL distribution 'Ubuntu-20.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 127.0.0.1 to reach the host.
WSL usbip: error: tcp connect
usbipd: error: Failed to attach device with busid '2-2'.

C:\Users\wm692\Desktop>wsl --distribution Ubuntu-20.04 --user root -- bash -c 'echo < /dev/tcp/127.0.0.1/3240'
The system cannot find the path specified.

wm692202103, 2023-12-11:
> C:\Users\wm692\Desktop>usbipd list Connected: BUSID VID:PID DEVICE STATE 2-2 10c4:ea60 CP2102 USB to UART Bridge Controller Shared 2-3 8087:0032 英特尔(R) 无线 Bluetooth(R) Not shared 2-4 13d3:56fb [...]

> C:\Users\wm692\Desktop>usbipd attach --wsl --busid 2-2 usbipd: info: Using WSL distribution 'Ubuntu-20.04' to attach; the device will be available in all WSL 2 distributions. usbipd: info: Using IP [...]

After the above operation on another device, it can be connected normally, but I don't know why this device cannot connect

dorssel, 2023-12-11:
OK, I'm sorry. You are using cmd.exe (I was assuming powershell). Try with double quotes:

```
wsl --distribution Ubuntu-20.04 --user root -- bash -c "echo < /dev/tcp/127.0.0.1/3240"
```

Marauders666, 2023-12-11:
Good afternoon Maybe off topic, it’s also worth setting up a network with a mirror. When I try to connect usb to wsl, the following error appears, the windows firewall is turned off, also for hyper-v:

2-13   0bc2:a003  Запоминающее USB-устройство SCSI (UAS)                        Shared

Persisted:
GUID                                  DEVICE

PS C:\Users\note> usbipd attach --wsl --busid=2-13
usbipd: info: Using WSL distribution 'Ubuntu-22.04' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 127.0.0.1 to reach the host.
usbipd: warning: A third-party firewall may be blocking the connection; ensure TCP port 3240 is allowed.

dorssel, 2023-12-11:
@Marauders666 
That's not an error, but a warning. Does the actual attach work?

Marauders666, 2023-12-11:
> @Marauders666 What is your output of:
> 
> ```
> wsl --distribution Ubuntu-22.04 --user root -- bash -c "time echo < /dev/tcp/127.0.0.1/3240"
> ```

Marauders666, 2023-12-11:
> @Marauders666 What is your output of:
> 
> ```
> wsl --distribution Ubuntu-22.04 --user root -- bash -c "time echo < /dev/tcp/127.0.0.1/3240"
> ```
There is no conclusion. I already entered this, it seems like at the end it just gave timeout

Marauders666, 2023-12-11:
> @Marauders666 What is your output of:
> 
> ```
> wsl --distribution Ubuntu-22.04 --user root -- bash -c "time echo < /dev/tcp/127.0.0.1/3240"
> ```

wsl --distribution Ubuntu-22.04 --user root -- bash -c "time echo < /dev/tcp/127.0.0.1/3240"
bash: connect: Connection timed out
bash: line 1: /dev/tcp/127.0.0.1/3240: Connection timed out

real    2m9.487s
user    0m0.001s
sys     0m0.000s

dorssel, 2023-12-11:
This is the problem. The `usbipd-win` service running on Windows cannot be connected to. Is it running (you can check in the services app)? It is (or should be) listening on port 3240. If it is running, there are firewalling issues.

wm692202103, 2023-12-11:
> OK, I'm sorry. You are using cmd.exe (I was assuming powershell). Try with double quotes:
> 
> ```
> wsl --distribution Ubuntu-20.04 --user root -- bash -c "echo < /dev/tcp/127.0.0.1/3240"
> ```

PS C:\Users\wm692\Desktop> wsl --distribution Ubuntu-20.04 --user root -- bash -c "echo < /dev/tcp/127.0.0.1/3240"
bash: connect: Connection refused
bash: /dev/tcp/127.0.0.1/3240: Connection refused

dorssel, 2023-12-11:
@wm692202103 
Are you sure the `usbipd-win` service is actually running? What is the output of `sc query usbipd`?

wm692202103, 2023-12-12:
> @wm692202103 Are you sure the `usbipd-win` service is actually running? What is the output of `sc query usbipd`?

Thank you for your enthusiastic reply. This issue has been resolved by reinstalling wsl2-ubuntu20.04.6

closed 2023-12-12 by dorssel

Marauders666, 2023-12-12:
> @wm692202103 Are you sure the `usbipd-win` service is actually running? What is the output of `sc query usbipd`?

I tried to reinstall the image like the person above, but it didn’t help (he also writes about the firewall

dorssel, 2023-12-12:
@Marauders666 

> bash: connect: Connection timed out
bash: line 1: /dev/tcp/127.0.0.1/3240: Connection timed out

This is your problem: in mirrored mode, WSL should be able to connect to the host using the localhost address. This is either some firewall issue, or something wrong with your installation of WSL. It is unrelated to `usbipd-win`.

Marauders666, 2023-12-12:
> @Marauders666
>
>
> This is your problem: in mirrored mode, WSL should be able to connect to the host using the localhost address. This is either some firewall issue, or something wrong with your [...]

I disabled all network connections so that only localhost remained. Firewall and Defender are disabled. I also tried it with the firewall turned on, same thing. I don't have any firewalls or antiviruses anymore.

Marauders666, 2023-12-12:
I also switched to nat mode and it failed to attach.

Marauders666, 2023-12-14:
Everything worked out. If you, like me, use an assembly (in my case ReviOs), the problem is that some service is missing or, on the contrary, some service is enabled. Update the system using the latest windows image