dorssel/usbipd-win #797: Cannot connect USB Storage device with new version of usbipd 4.0.0

piwaneczko, 2023-12-11:
Hi @dorssel. I have recompiled WSL kernel (5.15.133.1-microsoft-standard-WSL2+) with USB_STORAGE enabled. I was trying today to update your software to 4.0.0 and it seems to not work with USB storage devices. When I'm trying to connect my card reader device - `1-16  0bda:0306  Realtek USB 3.0 Card Reader` with command: `usbipd attach -w -b 1-16` i have following error message:
```
usbipd: info: Using WSL distribution 'Ubuntu' to attach; the device will be available in all WSL 2 distributions.
usbipd: info: Using IP address 172.23.16.1 to reach the host.
WSL usbip: error: Attach Request for 1-16 failed - Device in error state
usbipd: error: Failed to attach device with busid '1-16'.
```

When i reverted to version 3.20 everything works.

piwaneczko, 2023-12-11:
Hi. WTF?? :) I've tried again, and it works :). But I saw one difference during updating with `winget install usbipd`. Windows asked me to restart the system. By the way. Thanks for fast response, and great respect for your work.

dorssel, 2023-12-11:
Ah, the dreaded "have you tried a reboot" Windows solution for everything...
If your issue is resolved, please close.

closed 2023-12-11 by piwaneczko