dorssel/usbipd-win #798: Attachment surviving after reboot

Ehsan-Home, 2023-12-13:
Hi. Thanks for this great library!

I noticed in the documentation that it mentions: 

> Attaching devices to a client is non-persistent. You will have to re-attach after a reboot, or when the device resets or is physically unplugged/replugged.

Is there any method or workaround to establish persistent attachment? This would mean setting up the attachment once and having it remain available even after a reboot.

Thanks in advance :)

dorssel, 2023-12-13:
There is `--auto-attach`. And you could start such an auto-attach in a simple Windows startup script so it works right on boot.

Ehsan-Home, 2023-12-14:
@dorssel Thanks for your reply. 

I will leverage `--auto-attach` and write a script (.bat) format to automate this.

bimmerdriver, 2024-02-27:
I'm using uspipd-win with WSL / Ubuntu. I have a service that starts when linux boots and three other services depend on the first service starting properly. If the USB device isn't available, the first service doesn't start properly, causing problems for the other services. I have to stop them all and then restart them.

I just tried shutting down WSL, then running usbipd attach --wsl --busid=7-3 --auto-attach. It failed because WSL was not running. I booted WSL, then ran the attach command again and it worked. Either I'm confused about what this option is supposed to do or it's not working.

I need to have the USB device attached as soon as WSL boots. Is there a way to do this? Maybe systemd could be used if the device can be attached from WSL.

dorssel, 2024-02-27:
@bimmerdriver 
Attaching always has to be initiated by the *client* (WSL in your case). All that `usbipd attach --wsl` does is run some commands on WSL that you could have run yourself from WSL, as a "convenience macro". Of course, WSL needs to be running to do that. If you need to attach early in the boot procedure, you will have to setup a systemd service (like you did). You can do something similar as `auto-attach.sh` (which is the script that is part of `usbipd-win` and that is run on the WSL side by `usbipd attach --wsl --auto-attach`): try to attach in an endless loop.

bimmerdriver, 2024-02-28:
@dorssel Thank you for your reply. I haven't actually set up a systemd service, yet. This is not a "production" system and I'm not a linux expert, so I'm looking for a quick and dirty solution that's easy to set up even if it won't survive an upgrade. It looks like it's necessary to use systemd even to enable rc.local.

I know it's a gratuitous request and even taking care of a simple use-case isn't simple, but it would be great if this capability was part of a package. I realize others would have more complicated requirements, but I would be happy even if I had to edit a file to configure it.

dorssel, 2024-02-28:
@bimmerdriver 
I hate to disappoint you, but `usbipd-win` focusses on the Windows server side. Your issue really is a client-side issue. Yes, `usbipd-win` has some WSL support, but only the basics and always distribution independent. Setting up a service is *very* distribution dependent. Besides, WSL is only one type of client, `usbipd-win` isn't actually WSL specific. TL;DR: this is outside the scope of `usbipd-win`.